node-opcua/node-opcua #305: getMonitoredItems goes wrong with MicroProfile

zombinary, 2017-02-08:
I work with an opc Server with Micro Profile and could not subscribe on a variable.

The protoype 'getMonitoredItems()' returned an error which contains the response 'ServiceFault' with error code 'BadServiceUnsupported'.
This is an permissible error code, because the profile must not supported this service. 
(see also: http://opcfoundation-onlineapplications.org/profilereporting/)

I changed the prototype to tag this error, but i'm not sure to do this here or on a higher level (performMessageTransaction).

**client_session.js**
`/**
 * @method getMonitoredItems
 * @param subscriptionId {UInt32} the subscription Id to return
 * @param callback {Function}
 * @param callback.err {Error}
 * @param callback.monitoredItems the monitored Items
 * @param callback.monitoredItems the monitored Items
 */
ClientSession.prototype.getMonitoredItems = function (subscriptionId, callback) {

    // <UAObject NodeId="i=2253"  BrowseName="Server">
    // <UAMethod NodeId="i=11492" BrowseName="GetMonitoredItems" ParentNodeId="i=2253" MethodDeclarationId="i=11489">
    // <UAMethod NodeId="i=11489" BrowseName="GetMonitoredItems" ParentNodeId="i=2004">
    var self = this;
    var methodsToCall =
        new call_service.CallMethodRequest({
            objectId: coerceNodeId("ns=0;i=2253"),  // ObjectId.Server
            methodId: coerceNodeId("ns=0;i=11492"), // MethodIds.Server_GetMonitoredItems;
            inputArguments: [
                // BaseDataType
                {dataType: DataType.UInt32, value: subscriptionId}
            ]
        });

    self.call([methodsToCall], function (err, result, diagnosticInfo) {
    		/* istanbul ignore next */
        if (err) {
        	if(err.response.responseHeader.serviceResult.value === 0x800b0000 || err.response.responseHeader.serviceResult.value === 0x80400000){
        		/* micro profile return with 'BadServiceUnsopported'  || 'BadNotImplemented' */
        		var data = {
                serverHandles: null, //
                clientHandles: null
            };
            callback(null, data, diagnosticInfo);
            return
        	}else{
          	return callback(err);
        	}
        }else{
          result = result[0];
          diagnosticInfo = diagnosticInfo ? diagnosticInfo[0] : null;
          //xx console.log(" xxxxxxxxxxxxxxxxxx RRR err",err);
          //xx console.log(" xxxxxxxxxxxxxxxxxx RRR result ".red.bold,result.toString());
          //xx console.log(" xxxxxxxxxxxxxxxxxx RRR err",diagnosticInfo);
          if (result.statusCode !== StatusCodes.Good) {

              callback(new Error(result.statusCode.toString()), result, diagnosticInfo);

          } else {

              assert(result.outputArguments.length === 2);
              var data = {
                  serverHandles: result.outputArguments[0].value, //
                  clientHandles: result.outputArguments[1].value
              };
              assert(data.serverHandles instanceof Uint32Array);
              assert(data.clientHandles instanceof Uint32Array);
              callback(null, data, diagnosticInfo);
          }
        }
        }
    `

erossignon, 2017-02-10:
I am not sure what you want to achieve.
The call to getMonitoredItems returns an error as expected because the service is not implemented by the server,

Please tell me how the ClientSession#getMonitoredItems get called on your end ?

zombinary, 2017-02-10:
I would supscribe a variable from an server with Mirco Profile.
But thats goes wrong because the error 'BadServieUnsupported' returns.
        	
It was a problem in my function call and a wrong error handling
Now it works and everything is ok.
        	
THX for your answer.

closed 2017-02-10 by zombinary