node-opcua/node-opcua #308: Message from Client to Server

DynamicRemo, 2017-02-14:
From sample_server.js, sample_client.js tutorial this [Question](https://github.com/node-opcua/node-opcua/issues/134) explains the Message passing from Server to Cient. Thanks for that, but in my case i have multiple clients connected to single Server. How can i send message from Client to Server?

Thanking You!

erossignon, 2017-02-14:
Could you elaborate and provide a more precise description ?

DynamicRemo, 2017-02-16:
Simple scenario would be my RaspberryPi as a Client and my Laptop as a Server. I want RaspberryPi to act as a Node OPC UA Client and send the sensor values to my laptop, which is Node OPC UA Server.

We can send message from server to clients via Nodejs OPC UA and there are examples for that but how can i send message from Client to Server?

erossignon, 2017-03-09:
I would suggest that you inverse the role : usually,  the Rapsberry should read sensor values and expose them through the address space  of its opcua server, the opcua client shall be your laptop.

DynamicRemo, 2017-03-14:
Multiple Servers (RaspberryPi) are sending data to one single Client (my Laptop), wouldn't that be strange? 
Kindly, guide me accordingly. Thanking You!

erossignon, 2017-03-14:
This is how it will work in opcua.
* Each of the raspberry pi computer will run its own opcua server and expose its sensor data as opcua variable. 
* Your laptop will establish as many connections to each of your Raspberry PI servers and will subscribe to to all relevant  variables. This what we call monitored it's.S.
* From this point of onward, the server will publish any data change on the monitored variables back to the client automatically when they change.

DynamicRemo, 2017-03-14:
I was concerned with the traditional client-server communication, in which multiple clients are communicating/reporting to one single server. But yeah, i got your approach will try to proceed with that. Thank you!

closed 2017-03-31 by erossignon

aabm00, 2017-11-27:
Hi !!!

@erossignon I am just starting to look how this protocol works and your library looks great. I have been going through the examples to see how it works and in reference to this post I have some doubts:

I want to develop an application (let's call it **App1**) that listen for differents **devices** (arduino, raspberry,,) with differents kind of sensors. And at the same time I want that third party applications (let's call them **AppsThird**) could subcribe to **App1** in order to get data from this differents devices. 
I don't want that **AppsThird** can access directly to the **Devices**, for security and to make simple the access to of **AppsThird** to multiple devices with only one connection to **App1**.
You recomended to @DynamicRemo that the **devices** should be Servers and the **App1** the Client, but in my use case if **App1** is a Client 
How can connect the **AppsThird** with **App1**?
Can be stablished a Server to Server communication? So 
**Devices** --> OPCUA Server
**App1** --> OPCUA Server
**AppThird** --> OPCUA Clients

I hope my explanation is understandable

Thanks

allatrue, 2017-11-28:
Your App1 could be a server to AppThird and a client to Devices. This is like for example KepServer does, if you take an OPC-UA driver to access devices with OPC-UA server.

aabm00, 2017-11-28:
Thanks @allatrue  for you quick answer!!

It's make sense, but then if I have implemented in App1 an **OPC-UA Client** to subscribe to **Devices** and an **OPC-UA Server** to  publish  to **AppThird**.

How does the communication between **App1-OPC-UA Client** and **App1-OPC-UA Server** be to easily republish the **Devices-OPC-UA server** Variables in the **App1-OPC-UA Server**?

Thanks

allatrue, 2017-11-30:
OPC-UA is per design Client-Server. In your case a broker based architecture with pub/sub pattern is better approach. OPC-Foundation is working on OPC-UA pub/sub specification.

So either you implement a mapping between both parts of your App1 (forwarding subscriptions in one direction and notifications in another) or you find third-party OPC-UA gateway.

Other option ist to switch to another technology, that better fits your requirements, for example MQTT, WAMP, a so on.

aabm00, 2017-11-30:
Thanks @allatrue 

I think I am going to try a mix between MQTT & OPCUA.