node-opcua/node-opcua #309: Java OPC UA as a Server and Nodejs OPC UA as a Client

DynamicRemo, 2017-02-14:
Is this possible to have [Java OPC UA](https://github.com/OPCFoundation/UA-Java) as a Server and [Nodejs OPC UA](http://node-opcua.github.io/) as a Client? 

Just like websocket and message-queuing are supported in this cross platform communication manner, so far i am not able to find something similar in OPC UA.
Although, i tried connecting my Java implementation as a Server to my Nodejs implementation as a Client and they got connected but soon after that i got these error and still have no success in message passing.

`Connected at  opc.tcp://<MY PC NAME>.mshome.net:8666/UAExample

 WARNING SERVER responseHeader.requestHandle is invalid: expecting 0x3  but got 0x0  Class = ServiceFault
 failure  { Error:  ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2017-02-14T16:39:53.950Z
   requestHandle               /* IntegerId                        */: 0               0x0
   serviceResult               /* StatusCode                       */: BadInternalError (0x80020000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: 0
     additionalInfo            /* String                           */:  at org.opcfoundation.ua.application.ServiceHandlerComposition$1.serve(ServiceHandlerComposition.java:236)
        at org.opcfoundation.ua.application.ServiceHandlerComposition.serve(ServiceHandlerComposition.java:138)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerSecureChannel.handleSecureMessage(OpcTcpServerSecureChannel.java:113)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerConnection.handleSecureMessage(OpcTcpServerConnection.java:681)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerConnection$3.onMessageComplete(OpcTcpServerConnection.java:256)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder.fireComplete(SecureInputMessageBuilder.java:252)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder.setMessage(SecureInputMessageBuilder.java:278)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder$1.run(SecureInputMessageBuilder.java:140)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)

     innerStatusCode           /* StatusCode                       */: BadInternalError (0x80020000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
       namespaceUri            /* Int32                            */: -1
       symbolicId              /* Int32                            */: -1
       locale                  /* Int32                            */: -1
       localizedText           /* Int32                            */: 1
       additionalInfo          /* String                           */:  at org.opcfoundation.ua.examples.ServerExample1$MyServerExample.getServerSignature(ServerExample1.java:248)
        at org.opcfoundation.ua.examples.ServerExample1$MyServerExample.onCreateSession(ServerExample1.java:190)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at org.opcfoundation.ua.application.ServiceHandlerComposition$1.serve(ServiceHandlerComposition.java:225)
        at org.opcfoundation.ua.application.ServiceHandlerComposition.serve(ServiceHandlerComposition.java:138)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerSecureChannel.handleSecureMessage(OpcTcpServerSecureChannel.java:113)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerConnection.handleSecureMessage(OpcTcpServerConnection.java:681)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerConnection$3.onMessageComplete(OpcTcpServerConnection.java:256)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder.fireComplete(SecureInputMessageBuilder.java:252)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder.setMessage(SecureInputMessageBuilder.java:278)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder$1.run(SecureInputMessageBuilder.java:140)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)

       innerStatusCode         /* StatusCode                       */: BadInternalError (0x80020000)
       innerDiagnosticInfo     /* DiagnosticInfo                   */: {
       }
     }
   }
   stringTable                 /* String                        [] */: [ ServiceFault: Bad_InternalError (0x80020000) "An internal error occurred as a result of a programming or configuration error."
Diagnostic Info: java.lang.NullPointerException
                at org.opcfoundation.ua.examples.ServerExample1$MyServerExample.getServerSignature(ServerExample1.java:248)
        at org.opcfoundation.ua.examples.ServerExample1$MyServerExample.onCreateSession(ServerExample1.java:190)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at org.opcfoundation.ua.application.ServiceHandlerComposition$1.serve(ServiceHandlerComposition.java:225)
        at org.opcfoundation.ua.application.ServiceHandlerComposition.serve(ServiceHandlerComposition.java:138)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerSecureChannel.handleSecureMessage(OpcTcpServerSecureChannel.java:113)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerConnection.handleSecureMessage(OpcTcpServerConnection.java:681)
        at org.opcfoundation.ua.transport.tcp.nio.OpcTcpServerConnection$3.onMessageComplete(OpcTcpServerConnection.java:256)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder.fireComplete(SecureInputMessageBuilder.java:252)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder.setMessage(SecureInputMessageBuilder.java:278)
        at org.opcfoundation.ua.transport.tcp.nio.SecureInputMessageBuilder$1.run(SecureInputMessageBuilder.java:140)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)

java.lang.NullPointerException] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
    at process_request_callback (/home/pi/DevProjects/NodeOPCUA/myclient/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:66:15)
    at ClientSecureChannelLayer._on_message_received (/home/pi/DevProjects/NodeOPCUA/myclient/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:121:5)
    at emitMany (events.js:127:13)
    at MessageBuilder.emit (events.js:201:7)
    at MessageBuilder._decode_message_body (/home/pi/DevProjects/NodeOPCUA/myclient/node_modules/node-opcua/lib/misc/message_builder.js:445:22)
    at MessageBuilderBase._feed_messageChunk (/home/pi/DevProjects/NodeOPCUA/myclient/node_modules/node-opcua/lib/misc/message_builder_base.js:189:18)
    at PacketAssembler.<anonymous> (/home/pi/DevProjects/NodeOPCUA/myclient/node_modules/node-opcua/lib/misc/message_builder_base.js:49:14)
    at emitOne (events.js:96:13)
    at PacketAssembler.emit (events.js:188:7)
    at PacketAssembler.feed (/home/pi/DevProjects/NodeOPCUA/myclient/node_modules/node-opcua/lib/transport/packet_assembler.js:102:14)
  response:
   ServiceFault {
     responseHeader:
      ResponseHeader {
        timeStamp: 2017-02-14T16:39:53.950Z,
        requestHandle: 0,
        serviceResult: [Object],
        serviceDiagnostics: [Object],
        stringTable: [Object],
        additionalHeader: null } } }
`
 
Please guide, Thanking You!

lemme, 2017-03-17:
I have got a similar problem to DynamicRemo ...
... I use two raspberry pis and of course a PC
... on both raspberry pis there is javascript stack for opc ua (version 0.41) installed
... the "server" opc ua pi generates data like temperature
... on the PC I use the OPC UA Expert client to get these temperatures
... everything works fine! ->no fall out or crash
... now I take the second raspberry pi (jessie, node-red, javascript for opc ua version 0.41)
-> I can build a programm to show the temperature
-> I can deploy the programm 
-> I can see the gauge and it works -> BUT only for milliseconds until it stops ...
... the loglooks like this:

> temperature: 22
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> temperature: 23
> MessageBuilder : ERROR DETECTED IN event handler
> FILE - message_builder.js - LINE 452 : this indicates a bug in the code that uses this class
> TypeError: Cannot read property 'value' of null
> at ClientMonitoredItem.<anonymous> (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:536:61)
> at emitOne (events.js:96:13)
> at ClientMonitoredItem.emit (events.js:188:7)
> at ClientMonitoredItem._notify_value_change (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_monitored_item.js:102:10)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:189:32
> at Array.forEach (native)
> at ClientSubscription.__on_publish_response_DataChangeNotification (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:182:20)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:292:26
> at Array.forEach (native)
> at ClientSubscription.onNotificationMessage (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:287:26)
> at ClientSidePublishEngine._receive_publish_response (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_publish_engine.js:341:22)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_publish_engine.js:199:18
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_session.js:609:9
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_session.js:887:9
> at modified_callback (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:942:24)
> at process_request_callback (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:75:5)
> at ClientSecureChannelLayer._on_message_received (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:121:5)
> at emitMany (events.js:127:13)
> at MessageBuilder.emit (events.js:201:7)
> at MessageBuilder._decode_message_body (/home/pi/.node-red/node_modules/node-opcua/lib/misc/message_builder.js:445:22)
> at MessageBuilderBase._feed_messageChunk (/home/pi/.node-red/node_modules/node-opcua/lib/misc/message_builder_base.js:189:18)
> at PacketAssembler.<anonymous> (/home/pi/.node-red/node_modules/node-opcua/lib/misc/message_builder_base.js:49:14)
> TypeError: Cannot read property 'value' of null
> at ClientMonitoredItem.<anonymous> (/home/pi/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:536:61)
> at emitOne (events.js:96:13)
> at ClientMonitoredItem.emit (events.js:188:7)
> at ClientMonitoredItem._notify_value_change (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_monitored_item.js:102:10)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:189:32
> at Array.forEach (native)
> at ClientSubscription.__on_publish_response_DataChangeNotification (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:182:20)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:292:26
> at Array.forEach (native)
> at ClientSubscription.onNotificationMessage (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:287:26)
> at ClientSidePublishEngine._receive_publish_response (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_publish_engine.js:341:22)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_publish_engine.js:199:18
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_session.js:609:9
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_session.js:887:9
> at modified_callback (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:942:24)
> at process_request_callback (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:75:5)
> at ClientSecureChannelLayer._on_message_received (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:121:5)
> at emitMany (events.js:127:13)
> at MessageBuilder.emit (events.js:201:7)
> at MessageBuilder._decode_message_body (/home/pi/.node-red/node_modules/node-opcua/lib/misc/message_builder.js:445:22)
> at MessageBuilderBase._feed_messageChunk (/home/pi/.node-red/node_modules/node-opcua/lib/misc/message_builder_base.js:189:18)
[... 84 lines cut ...]
> at ClientMonitoredItem._notify_value_change (/home/pi/.node-red/node_modules/node-opcua/lib/client/client_monitored_item.js:102:10)
> at /home/pi/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:189:32
> at Array.forEach (native)
[... cut ...]
> at PacketAssembler.<anonymous> (/home/pi/.node-red/node_modules/node-opcua/lib/misc/message_builder_base.js:49:14)

then it stops and no changes of the temperature were shown (log, ui)?
There are changes because I can check this out via OPC UA expert on the PC, where everthing works fine?
=> So as the comment of the message_builder.js part says => _this indicates a bug in the code that uses this class_
=> is there someone who can help, because I started with node-red 4 days ago ;-) ? 
--------------------------------------------

erossignon, 2017-03-18:
version (version 0.41) of the javascript stack is outdated, please upgrade to the latest ( 0.0.61) or above

closed 2017-10-07 by erossignon