node-opcua/node-opcua #312: Mandatory Placeholder causes server.initialize to fail

tomsoftware, 2017-02-20:
The call of server.initialize(post_initialize); does not raise an error neither calls the callback "post_initialize" if the nodeset_filename: [ path.join(modelFolder, "errortest.xml") ] contains an MandatoryPlaceholder Type?

    "use strict";

    var opcua = require("node-opcua");
    var path = require("path");

    var rootFolder = path.join(__dirname, "../");
    var modelFolder = path.join(rootFolder, "modeling/");

    // Let's create an instance of OPCUAServer
    var server = new opcua.OPCUAServer({
        port: 26543, // the port of the listening socket of the server
        resourcePath: "UA/MyLittleServer", // this path will be added to the endpoint resource name
        nodeset_filename: [
            opcua.standard_nodeset_file,
            path.join(modelFolder, "errortest.xml")
        ]
    });

    function post_initialize() {
        console.log("test"); /// <- never called
    }

    server.initialize(post_initialize);

[image]

erossignon, 2017-03-16:
there is a typo in the XML file that cause the xmlparser to get stuck
```<UAObject... BrowseName="1:&lt;values>">```
shall be
```<UAObject ... BrowseName="1:&lt;values&gt;"> ```
Took me a while to find !

tomsoftware, 2017-03-20:
Thank you for looking at this bug but I'm not sure if this is really a typo... I think it is valid xml (but of cause bad style :-( )... I test this with Firefox, Edge and Chrome. They all can pars this. I create this with the UaModeler so other people may have the same issue - UaModeler is creating this "style" on every "<>" appearance in a node/browse name :-( ?!

- bad style position 1:

      BrowseName="1:&lt;values>"

- bad style position 2:

      <DisplayName>&lt;values></DisplayName>

now I fix this for my by search & replace the 

    >">
with

    &gt;">

sorry for ruin your weekend...

[original xml error_xml.zip](https://github.com/node-opcua/node-opcua/files/853676/error_xml.zip)

closed 2018-01-28 by erossignon