node-opcua/node-opcua #313: Wrong nodeId second time causes uncaught exception

mikakaraila, 2017-02-21:
21 Feb 21:18:13 - [info] [OpcUa-Client:SUBSCRIBE] terminated monitoredItem on ns=4;s=Test32;datatype=Int32
21 Feb 21:18:13 - [error] [OpcUa-Client:SUBSCRIBE] subscription.monitorItem:Error: _socket has been disconnected by third party
warning : disconnection : closing pending sessions
 ClientTCP_transport Socket Error This socket has been ended by the other party
 Session is being transferred from channel 2  to channel  4
err =  null
21 Feb 21:18:15 - [info] [OpcUa-Client:OPC Read] Action on input:read Item from Topic: ns=5;s=Sinusoid1;datatype=Double session Id: ns=1;g=3032241F-B9DE-4BEE-9328-DD2C19687132
21 Feb 21:18:15 - [info] [OpcUa-Client:OPC Read] reading
21 Feb 21:18:15 - [info] [OpcUa-Client:OPC Read] {"topic":"ns=5;s=Sinusoid1","payload":"","_msgid":"85bfd302.1c752","datatype":"Double"}
21 Feb 21:18:15 - [info] [OpcUa-Client:OPC Read] Client status: active reading
21 Feb 21:18:16 - [info] [OpcUa-Client:OPC Read]        Node : ns=5;s=Sinusoid1
21 Feb 21:18:16 - [info] [OpcUa-Client:OPC Read]        Value : -0.03476497035177525
21 Feb 21:18:16 - [info] [OpcUa-Client:OPC Read]        DataType: 11 (Double)
21 Feb 21:18:16 - [info] [OpcUa-Client:OPC Read]        Message: ns=5;s=Sinusoid1 (Double)
21 Feb 21:18:16 - [info] [OpcUa-Client:OPC Read]        Status-Code:Good (0x00000)
21 Feb 21:18:20 - [info] [OpcUa-Client:OPC Read] Action on input:read Item from Topic: ns=5;s=Sinusoid1;datatype=Double session Id: ns=1;g=3032241F-B9DE-4BEE-9328-DD2C19687132
21 Feb 21:18:20 - [info] [OpcUa-Client:OPC Read] reading
21 Feb 21:18:20 - [info] [OpcUa-Client:OPC Read] {"topic":"ns=5;s=Sinusoid1","payload":"","_msgid":"f396a86.84e0258","datatype":"Double"}
21 Feb 21:18:20 - [info] [OpcUa-Client:OPC Read] Client status: active reading
21 Feb 21:18:20 - [info] [OpcUa-Client:OPC Read]        Node : ns=5;s=Sinusoid1
21 Feb 21:18:21 - [info] [OpcUa-Client:OPC Read]        Value : -1.7491713997698481
21 Feb 21:18:21 - [info] [OpcUa-Client:OPC Read]        DataType: 11 (Double)
21 Feb 21:18:21 - [info] [OpcUa-Client:OPC Read]        Message: ns=5;s=Sinusoid1 (Double)
21 Feb 21:18:21 - [info] [OpcUa-Client:OPC Read]        Status-Code:Good (0x00000)
21 Feb 21:18:25 - [info] [OpcUa-Client:OPC Read] Action on input:read Item from Topic: ns=5;s=Sinusoid1;datatype=Double session Id: ns=1;g=3032241F-B9DE-4BEE-9328-DD2C19687132
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read] reading
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read] {"topic":"ns=5;s=Sinusoid1","payload":"","_msgid":"a895f3b9.3f237","datatype":"Double"}
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read] Client status: active reading
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read]        Node : ns=5;s=Sinusoid1
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read]        Value : -1.7146221370943593
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read]        DataType: 11 (Double)
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read]        Message: ns=5;s=Sinusoid1 (Double)
21 Feb 21:18:26 - [info] [OpcUa-Client:OPC Read]        Status-Code:Good (0x00000)
21 Feb 21:18:29 - [info] [OpcUa-Client:SUBSCRIBE] Action on input:subscribe Item from Topic: ns=4;s=Test32;datatype=Int32 session Id: ns=1;g=6853353B-13B0-67E7-44D3-725DB3C561D7
21 Feb 21:18:29 - [info] [OpcUa-Client:SUBSCRIBE] subscribing
21 Feb 21:18:29 - [info] [OpcUa-Client:SUBSCRIBE] Client status: active subscribing
21 Feb 21:18:29 - [info] [OpcUa-Client:SUBSCRIBE] Session subscriptionId: 1
21 Feb 21:18:29 - [info] [OpcUa-Client:SUBSCRIBE] ns=4;s=Test32;datatype=Int32 samplingInterval 100
21 Feb 21:18:29 - [warn] [OpcUa-Client:SUBSCRIBE] SUBSCRIBE: Monitoring Event: ns=4;s=Test32;datatype=Int32 by interval of 100 ms
21 Feb 21:18:29 - [red] Uncaught Exception:
21 Feb 21:18:29 - TypeError: Cannot read property 'createMonitoredItems' of null
    at ClientMonitoredItem._monitor (C:\users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua\lib\client\client_monitored_item.js:182:30)
    at Immediate.wait_for_subscription_and_monitor (C:\users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua\lib\client\client_subscription.js:539:27)
    at runCallback (timers.js:637:20)
    at tryOnImmediate (timers.js:610:5)
    at processImmediate [as _immediateCallback] (timers.js:582:5)

erossignon, 2017-02-21:
related to https://github.com/mikakaraila/node-red-contrib-opcua/issues/30 ?

mikakaraila, 2017-02-22:
> related to mikakaraila/node-red-contrib-opcua#30

Yes

erossignon, 2017-04-01:
Closing as unit test in node-opcua didn't show any issue,and because related node-red-contrib-opcua issue is now closed.

closed 2017-04-01 by erossignon