node-opcua/node-opcua #314: Uninitialized variables when using UAModeler and loading an XML

AlexPeitz, 2017-02-28:
Hello Etienne,

I encountered a problem using the UAModeler and the node-opcua simple_server.js.

I tried to create an OPC UA Model using the Modeler, where I define an object of a machine which contains different variables Weight, Length, Height etc... These are Analog Values, they contain a value, an Unit and a Range. Now when I load my generated xml file, the server structure gets created just fine, the types and objects are fine etc.
The only problem I have is that no variable gets instantiated. Upon reading the Variable using the UAExpert, i get an error code "BadWaitingForInitialData (0x80320000)". I understand that you can bind Variables created using the modeler to Variants you create in code. Is this for the time being the only way to do so? Because even the EURange or EngineeringUnit Variables of AnalogItems do not get initialized, and it rapidly gets messy binding all this by code. 

Below I included some screenshots and code snippets for you to see. I would appreciate some help or information, thank you.

*XML File snippet for the Variable:*

```xml
<UAVariable DataType="EUInformation" ParentNodeId="ns=1;i=6037" NodeId="ns=1;i=6038" BrowseName="EngineeringUnits">
        <DisplayName>EngineeringUnits</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=6037</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ExtensionObject>
                <uax:TypeId>
                    <uax:Identifier>i=888</uax:Identifier>
                </uax:TypeId>
                <uax:Body>
                    <uax:EUInformation><uax:NamespaceUri>http://www.opcfoundation.org/UA/units/un/cefact</uax:NamespaceUri>
                        <uax:UnitId>5066068</uax:UnitId>
                        <uax:DisplayName>
                            <uax:Text>mm</uax:Text>
                        </uax:DisplayName>
                        <uax:Description>
                            <uax:Locale>meter</uax:Locale>
                            <uax:Text>millimetre</uax:Text>
                        </uax:Description>
                    </uax:EUInformation>
                </uax:Body>
            </uax:ExtensionObject>
        </Value>
    </UAVariable>
```

*Code snippet for loading the xml file:*

```javascript
var server_options = {

    certificateFile: server_certificate_file,
    privateKeyFile: server_certificate_privatekey_file,

    port: port,

    maxAllowedSessionNumber: maxAllowedSessionNumber,
    maxConnectionsPerEndpoint: maxConnectionsPerEndpoint,

    nodeset_filename: [
        standard_nodeset_file,
        path.join(__dirname,"/model/virtualproductionstation1.xml/")
    ],

    serverInfo: {
        applicationUri: makeApplicationUrn(get_fully_qualified_domain_name(), "NodeOPCUA-Server"),
        productUri: "NodeOPCUA-Server",
        applicationName: {text: "NodeOPCUA" ,locale:"en"},
        gatewayServerUri: null,
        discoveryProfileUri: null,
        discoveryUrls: []
    },
    buildInfo: {
        buildNumber: "1234"
    },
    serverCapabilities: {
        operationLimits: {
            maxNodesPerRead: 1000,
            maxNodesPerBrowse: 2000
        }
    },
    userManager: userManager,
    
    isAuditing: false
};

server_options.alternateHostname = argv.alternateHostname;

var server = new OPCUAServer(server_options);
```

*Screenshot of UaExpert reading the variable*

[image]

AlexPeitz, 2017-02-28:
Okay I was being very stupid and am sorry for that, just close this issue Etienne. For everyone that wants to know, you have to not only copy the xml file but also the .ua file to your directory.

AlexPeitz, 2017-03-16:
Hey Etienne, 

I extended your xml parser so that it supports parsing of AnalogItems containing EUInformation and Range properties, UInt32 and NodeId values and simple LocalizedText. Are you interested in receiving the code?

erossignon, 2017-03-16:
Yes please,  let's make a pull request, don't forget to add a dedicated
unit test associated with the new code.

jupiterbak, 2017-08-26:
Hey Alex, Hey Etienne,

i am facing the same issues. Is it possible to have the extension of the code that supports AnalogItemTypes? It will be great if it is available.

AlexPeitz, 2017-08-28:
Hey jupiterbak,

If I remember correctly, you need to replace your load_nodeset2.js with this one. It's located in node-modules/node-opcua/lib/address_space

Sorry Etienne I totally forgot about doing a pull request. Since I never contributed to github before, I'm not sure what you mean by dedicated unit test? 

[load_nodeset2.zip](https://github.com/node-opcua/node-opcua/files/1256409/load_nodeset2.zip)

erossignon, 2017-10-11:
@AlexPeitz  can you confirm that node-opcua@0.1.0-8 now load nodeset files with variables values (i.e analogItems etc ...) as expected ? then close the issue.

closed 2018-01-08 by erossignon

BrotKolben, 2018-03-06:
Was the fix for AnalogItems only? Im having issues with Double variables...

erossignon, 2018-03-06:
@BrotKolben  can you provide a insight of how the address space has been constructed on your end ? A piece of self contained source code will be useful.

BrotKolben, 2018-03-07:
@erossignon [https://drive.google.com/open?id=1CXme_9baftDDIhuFBqo2fH2sQogrAu68](url)
In this example I added a new Type "MyType" to the address space containing a double variable.

BrotKolben, 2018-03-20:
Sorry, link was not working, but should work now!