node-opcua/node-opcua #316: How to define user specific conditionName/conditionClassName for an AlarmConditionEvent?

TorstenKo, 2017-03-07:
I like to distinguish alarm conditions in different conditions class types:
E.g. Alarm 1 is from the condition class type "ProcessConditionClassType"
E.g. Alarm 2 is a from the condition class type "MaintenanceConditionClassType"
 
I tried to setup this by using the  conditionClassId in the **instantiateAlarmCondition** to define a condition specific class. But it was not working.

I tried to set the information in the method "**raiseNewCondition**"
"alarmcondition.raiseNewCondition(({
        message: {
            locale: stdlocale,
            text: alarmnode.description.text,
        },
        severity: 100,
        quality: opcua.StatusCodes.Good,
        conditionClassname: "ProcessConditionClassType",
        conditionName: "thats a Condition Name"
    }));

I always get as "conditionClassName" and "conditionName": "Test". Is this not implemented? Is there any way to define a user specific conditionClassName and conditionName for a Condition?

erossignon, 2017-03-09:
This is now fully implemented, you can add conditionClass and conditionName options when calling instantiateAlarmCondition:
example:

```javascript
            var condition = addressSpace.instantiateCondition("AlarmConditionType",{
                 componentOf: source,
                 conditionSource: source,
                 browseName: "AlarmCondition1",
 
                 conditionClass: "ProcessConditionClassType",
                 conditionName: "MyConditionName"
             });
```

closed 2017-03-09 by erossignon

TorstenKo, 2017-03-10:
Thanks, its working now. Is there any way to predefine the severity of a condition.

erossignon, 2017-03-11:
This will do:
```
 condition.currentBranch().setSeverity(100);
````