node-opcua/node-opcua #319: How to create server with username and password authentication?

CrNix, 2017-03-08:
Hi,
is it possible to create server that allows to connect clients only with given username and password?
I can't find it in the code but I could use this feature.

Thank you

marcelwinh, 2017-03-09:
Hi,
there is an example in the [simple_server.js](https://github.com/node-opcua/node-opcua/blob/master/bin/simple_server.js)

```
var userManager = {
    isValidUser: function (userName, password) {
        if (userName === "user1" && password === "password1") {
            return true;
        }
        if (userName === "user2" && password === "password2") {
            return true;
        }
        return false;
    }
};

var server_options = {
	...
	userManager: userManager,
    
	allowAnonymous: false,
};

var server = new OPCUAServer(server_options);
....
```

CrNix, 2017-03-10:
It works!
Thank you very much

closed 2017-03-10 by CrNix

yairbloom, 2022-08-05:
Could you send a reference for example using "isValidUserAsync"?

Many thanks

muneebshafiq, 2023-09-01:
Hey @marcelwinh 

Can you help with the implementation with isValidUserAsync as it doesn't return anything so there's a timeout at client side. 
I'm implementing it like this but it isn't working.

`userManager: {
                isValidUserAsync: async function (username: string, password: string) {
                    const userInfo = await getUserByCredentials(username, password)
                    if (userInfo) {
                    console.log(userInfo, username, password);
                        return true
                    }
                    else {
                        return false;
                    }
                }
            },
allowAnonymous: false`

kevin-heneghan-tfs, 2023-10-23:
How about an example using async to an authentication server?  I make the authentication call successfully, but the server is not waiting on the response from the async handler?

`    var userManager = {
      isValidUserAsync: async function (userName, password) {
        let body = { username: userName, password: password}
        const response = await makeAuthServerPostRequest("/user/login", body);
        log.warn("**************************************************************");
        if (response.status == 200) {
          log.info("Authenticating user " + userName + " result " + JSON.stringify(response.data));
          if (response.data && response.data.token) {
            log.info("Authenticating user " + userName + " is valid");
            return true;
          }
        }
        return false;
      }
    };`