node-opcua/node-opcua #320: Dealing with structures

rauschh, 2017-03-09:
What is best practice for readig/writing and monitoring arbitrary structures consisting of elementary types, arrays and substructures?

Are there any examples showing this? 

Thanks in advance

erossignon, 2017-03-10:
If you means _structure_ = "OPCUA ExtensionObject" 
 * This is not easily supported at this time for structure that are not from namespace 0( standard OPCUA namespace) I am afraid.  

* Your use case could be of interest to progress on this. Can you provide a more detailed scenario ?

rauschh, 2017-03-10:
Yes, I am talking about extension objecst. In my IEC61131 PLC there are several instances of the same structure ([image]), which are made available to the OPC UA server. I can see them in UAExpert as ExtensionObjects (
[image])

It would be really nice to read/write and monitor them easily. In case of monitoring the changed event should fire as soon as at least one sub element has changed. Actually the client sample code is doing exactly this, but also throws an  "#getConstructor : cannot find constructor for expandedId" message. 
Obviously the SDK doesn´t know about the type´s structure, which - as it seems - is provided by the server as well. 

As a kind of workaround it would be OK for me to receive and provide raw data of the extension object as byte array or stream in order to parse it on my own. 

Any ideas?

Thanks for your efforts.

georglang, 2017-07-19:
@rauschh have you found a workaround to solve this issue ?
@erossignon are you working on this issue ?

nospam2000, 2017-07-24:
I am also very interested in the support of ExtensionObjects.

swtcwj, 2017-10-03:
Hi,
I´m currently running a project where the use of structured types would be highly interesting. I need to send PLC values to the OPC UA server, where a { ValueID, Value, timestamp } structure would be very useful in order to have client-side timestamping available within the dataset.

Thank you for any comments on this!

kikolehm, 2017-11-23:
Is there way to convert structure at least to buffer and provide that on data change? In addition is there some way to update all values at once for structure e.g. providing some same kind of json structure or so?

teropes, 2018-04-19:
Hi,
I believe in any real life use scenarios there are almost always PLC `struct` types (aka ExtensionObjects in OPC UA) being used and that requires custom types to be used. Some very simple bit poller type of applications might not need it, but otherwise it should be very core functionalities of the framework.

For our real world project needs I have developed the small functionality `opcua.types.js`, which registers custom structured data types to the node-opcua client or server and then allows us to read / write those custom types. It has been used in real application running against Siemens S7-1500 PLC's and it has been working quite ok.

This code is reverse engineered from the frameworks way of registering built-in structured types:
```
"use strict";

var assert = require("node-opcua-assert");
var util = require("util");

var schema_helpers = require("node-opcua-factory/src/factories_schema_helpers");
var ec = require("node-opcua-basic-types");
var register_class_definition = require("node-opcua-factory/src/factories_factories").register_class_definition;

var BaseUAObject = require("node-opcua-factory/src/factories_baseobject").BaseUAObject;
var ExpandedNodeId = require("node-opcua-nodeid/src/expanded_nodeid").ExpandedNodeId;
var NodeIdType = require("node-opcua-nodeid").NodeIdType;

// initialize map of constructors, returned after registration
const typeConstructors = [];

module.exports = function(typeSchemas) {
  typeSchemas.forEach(function(schema) {
    // define constructor function for custom type
    let TypeConstructor = function(options) {
      var self = this;
      options = options || {};

      assert(this instanceof BaseUAObject); //  ' keyword "new" is required for constructor call')

      schema_helpers.resolve_schema_field_types(schema);

      // call BaseUAObject constructor
      BaseUAObject.call(this, options);

      // initialize fields
      schema.fields.forEach(function(field) {
        if(field.fieldType === "ExtensionObject") {
          // field is extension object (nested structure) -> build instance of it
          self[field.propertyName] = new typeConstructors[field.constructor](options[field.propertyName]);
        } else {
          // field is basic data type
          self[field.propertyName] = schema_helpers.initialize_field(field, options[field.propertyName]);
        }
      });
    };

    // extend BaseUAObject
    util.inherits(TypeConstructor, BaseUAObject);

    // parse namespace and nodeId from full id string
    var namespace = schema.id.split(/(?!ns=)(\d+)/g)[1];
    var nodeId = schema.id.split(/(;s=|;i=)(.*)/g)[2];

    TypeConstructor.prototype.encodingDefaultBinary = new ExpandedNodeId(NodeIdType.STRING, nodeId, namespace);
    TypeConstructor.prototype._schema = schema;

    TypeConstructor.prototype.encode = function(stream, options) {
      var self = this;

      // call base class implementation first
      BaseUAObject.prototype.encode.call(self, stream, options);

      // encode fields
      schema.fields.forEach(function(field) {
        if(field.fieldType === "ExtensionObject") {
          // use extension objects encoder
          self[field.propertyName].encode(stream, options);
[... 114 lines cut ...]

      // prepare write options for client
      const writeOptions = {
        dataType: opcua.DataType[dataType],
        value: value
      };

      // write value
      self.opcSession.writeSingleNode(nodeId, writeOptions, function(err, results) {
        if (err || results.name !== "Good") return reject(err || results);
        resolve(results);
      });
    });
  }
```

Pros:
* it doesnt require patching the framework code, we have standard npm installation of node-opcua and this works on top of it
* supports also nesting of types, ExtensionObjects inside ExtensionObjects, that is also very common real life scenario, PLC structs can be organized in any way basically

Cons:
* you need to explicitly define your data types and for example field types and names must be correct exactly, otherwise encoding / decoding fails, but this is ok compromise to get it working
* client side works ok, but we have not yet developed it further on local development server side to get it actually working as a type, not just variable added to address space. Now it does not work for example with UAExpert correctly which would be massive help to testing / development, however client reading / writing against dev server still works.

Hopefully it helps some other users working with similar problems. Perhaps @erossignon also have a look and think about it if this type of functionality could be included in the node-opcua package. For example when creating the client instance, you could pass custom data types with the config and it would make constructors available for those.

nospam2000, 2018-04-19:
@teropes: Thank you so much, this is really great!

Just one question: do I have to define all fields of a structure or can I just choose the ones I need, e.g. omit "TaskCompleted" from your example?

  Michael

teropes, 2018-04-20:
@nospam2000 Yes you need to define the exact struct that the server has, otherwise there will be errors when encoding / decoding. If you have lot of spare bits / words in between, it makes the data type definitions a bit verbose, but it is required.

I'm thinking it could be easily improved if you want to ignore some fields, you could set it's `propertyName` to null and it would be ignored in the js object you handle, but would be correctly handled when encoding / decoding the object under the hood. Note: it does not happen now like that.

Chrissie, 2018-08-21:
@teropes I really, really appreciate the work you have done so far on this subject. With your code I was able to implement my own user defined type. 👍 
The only problem I am encountering at the moment is how to put a udt or udt array inside another udt. 
My two questions:
1: How to put a udt array inside another udt in the config file?
2: 
`
// initialize fields
schema.fields.forEach(function(field) {
    if(field.fieldType === "ExtensionObject") {
          // field is extension object (nested structure) -> build instance of it
          self[field.propertyName] = new typeConstructors[field.constructor](options[field.propertyName]);
        } else {
          // field is basic data type
          self[field.propertyName] = schema_helpers.initialize_field(field, options[field.propertyName]);
        }
});
`
Does this code from your first code sample have something to do with my problems? I do not see any field.constructor inside your types config?

Chrissie, 2018-08-22:
@teropes Thank you for your amazing work so far!

At the moment I am able to create and read/write user-defined types in and from OPCUA. The only problem I am encountering at the moment is the use of PLC Structs and PLC Struct arrays inside the user-defined type.

Does this have something to do with the field.constructor in the example code you gave? If so, how to put that constructor in the example config file? I do not see any constructor field there.

Code I referenced to:
```
      // initialize fields
      schema.fields.forEach(function(field) {
        if(field.fieldType === "ExtensionObject") {
          // field is extension object (nested structure) -> build instance of it
          self[field.propertyName] = new typeConstructors[field.constructor](options[field.propertyName]);
        } else {
          // field is basic data type
          self[field.propertyName] = schema_helpers.initialize_field(field, options[field.propertyName]);
        }
      });
    };
```

Thanks in advance!

mdameda, 2018-11-27:
> ```
> We also have a little wrapper for opcua client which makes `read`, `write`, `subscription` etc. friendlier to use, here is how `write` is implemented:
> 
> ```

I see the write method ... but how do you implemented  'read' of struct node ?

msalafia, 2019-02-28:
But the if the namespaceindex of the DefaultBinary_Encoding Object for my custom DataTypes is selected at runtime, how is possible to put it hardcoded in the schema?

@teropes In your example, you wrote `id: 'ns=3;s=TE_"TC_Communication"."TransferTask"'` but who assures you that the namespace index is 3?

teropes, 2019-03-01:
@msalafia I have used this now in several projects which all used Siemens S7-1500 series plc's and all of those had the namespace id of 3 used. It seems to be quite fixed, i have never seen it use any other one. This of course can be different with other devices / software opc ua servers.

We even looked at the option of programmatically generating our opc ua signal config from the xml file that Siemens TIA portal can export, but never implemented that.

There could also be some way of resolving node ids by signal name once on startup.

Btw this registered types thing was implemented against very old node-opcua version, it most likely breaks with more recent versions. I should dev it up to date. Even better would be having this functionality baked into node-opcua.

S-te-fan97, 2019-11-07:
@teropes @erossignon how can I use the reading and writing of an ExtensionObject in a node in Node-Red?

efeozazar, 2020-01-30:
This feature is a must have for this library in my opinion. I have used it in a project and got stuck in this part. Because of it need to write it again with another library and in another language. Very frustrating for me. Hope it will be available soon.

erossignon, 2020-01-31:
@efeozazar Please consider applying as a sponsor https://github.com/sponsors/node-opcua , this will accelerate the development of ExtensionObject support.

200Matteo, 2020-02-14:
I used what @teropes suggested and I can read ExtensionObject, thank you for your code.
I tried to write an Array of ExtensionObject using the same code as well, but when I create an instance of custom type, I got the error  
`UnhandledPromiseRejectionWarning: TypeError: Class constructor BaseUAObject cannot be invoked without 'new'` . 
There are a lot of issues on the web regarding this error, but no one help me to solve this problem. 
Is there another way or an updated code to do the call  `BaseUAObject.call(this, options);` that is done in `opcua.types.js`? 
Thank you!

mdameda, 2020-02-23:
> Hi,
> I believe in any real life use scenarios there are almost always PLC `struct` types (aka ExtensionObjects in OPC UA) being used and that requires custom types to be used. Some very simple bit [...]

Hi all, as @Chrissie wrote :

> The only problem I am encountering at the moment is the use of user-defined types and user-defined type arrays inside the user-defined 

For this reason I made a change to the library and also to the `opcua.types.js` file in order to manage this type of situations, and manage arrays of Int or arrays of Float.

### FIRST STEP
**opcua.types.js** 
The decode method is modified like so:

```
TypeConstructor.prototype.decode = function(stream, options) {
      var self = this;
      // call base class implementation first
      BaseUAObject.prototype.decode.call(this, stream, options);

      // decode fields
      schema.fields.forEach(function(field) {
        if(field.fieldType === "ExtensionObject") {
          // use extension objects decoder
          self[field.propertyName].decode(stream, options);
        } else {
          // use built-in data type decoder
          if (field.fieldType == 'ArrayFloat') {
            self[field.propertyName] = ec["decodeArray"](stream, function(stream) {
              var r = stream.readFloat();
              return r;
            });
          }
          else if (field.fieldType == 'ArrayInt16') {
            self[field.propertyName] = ec["decodeArray"](stream, function(stream) {
              var r = stream.readInt16();
              return r;
            });
          }
          else {
            self[field.propertyName] = ec["decode" + field.fieldType](stream, options);
          }
        }
      });
    };

```

I introduced two new types : **ArrayInt16**  and **ArrayFloat**

### SECOND STEP
modified
**factories_builtin_types.js**  into the module node-opcua-factory

I added two new rows into the  __defaultType_ array:

```
{name: "ArrayFloat", encode: ec.encodeArray, decode: ec.decodeArray, defaultValue: [], coerce: ec.coerceFloat}, 

{name: "ArrayInt16", encode: ec.encodeArray, decode: ec.decodeArray, defaultValue: [], coerce: ec.coerceInt16}
```

### THIRD STEP

My example of OpcUa custom structures mapping:

```

"use strict";

module.exports = [
{
		  "id":"ns=3;s=TE_\"CommandCode\"",
		  "name":"Command",
		  "fields":[
                        {
				"name":"ID_Command",
				"propertyName":"id",
				"fieldType":"UInt16"
			 },{
				"name":"Conf_Base",
				"propertyName":"confBase",
				"fieldType":"ArrayFloat"
			 }
                    ]
}]
```

This is a rough work , but could be a start point and i hope it can be useful to someone.
Enjoy

erossignon, 2020-03-20:
📢 This feature is now builtin node-opcua@2.5.1 !

* Extension Object are automatically decoded/encoded 
* supported in Server and client code

🌹 I would like to thank our sponsors for making this happen ! 👍

HelloHungryImDad, 2020-03-26:
@erossignon  thank you for implementing this functionality. I think this will make a lot of people happy. 

Can the encode feature be used to prepare and send a struct (in my case ns=1;i=3010 ScanSettings) as an argument to a method call? I've been struggling to understand how to do this for the last day or two. Is there a code snippet in the repo or one that you could post on here to help demonstrate how to use this feature please?

AlexGodbehere, 2020-04-24:
I'm also facing the same issue.

Looking at the Wireshark dump of what UAExpert is sending for the inputArguments of this method call, I only need to provide the ExpandedNodeId, EncodingMask, and ByteString for the method.

Is there a high-level function available like new ExtensionObject(ExpandedNodeId, EncodingMask, ByteString)?

erossignon, 2020-04-27:
yes, 
```
const extension object = session.createExtensionObject(dataTypeNodeId, { /* pojo */});
```
will help you create a extension object matching a structure dataType

HelloHungryImDad, 2020-04-27:
I'm trying to use this method without success. The client code I have:

```javascript
(async () => {

    const client = OPCUAClient.create({ endpoint_must_exist: false});
    client.on("backoff", () => console.log("Backoff: trying to connect to ", endpointUri));

    await client.withSessionAsync(endpointUri, async (session) => {
        let scanSettings = {
            Duration: 0,
            Cyles: 0,
            DataAvailble: false
        };
        const nodeID = new NodeId(NodeIdType.STRING, "rfr310.ScanStart.InputArguments", 4);
        const extObj = session.constructExtensionObject(nodeID, scanSettings);

        const methodsToCall = [
            {
                objectId: "ns=4;s=rfr310",
                methodId: "ns=4;s=rfr310.ScanStart",
                inputArguments: [extObj]
            }
        ];

        extObj.then(() => {
            session.call(methodsToCall,(err,results) => {
                if (err) {
                    console.log(err);
                } else {
                    console.log(results);
                }
            });
        }).catch(() => {
        })
    });
})();```

produces the error "dispose when pendingTransactions is not empty". 

What am I doing wrong?

HelloHungryImDad, 2020-04-29:
Here's a complete code snippet of how I managed to use it:

`
const { OPCUAClient, NodeId, NodeIdType, DataType} = require("node-opcua");
​
const endpointUri = "opc.tcp://Apps-mica-np63i:4840";
​
(async () => {
​
    const client = OPCUAClient.create({ endpoint_must_exist: false});
    client.on("backoff", () => console.log("Backoff: trying to connect to ", endpointUri));
​
    await client.withSessionAsync(endpointUri, async (session) => {
        // Scan settings value input
        const scanSettingsParams = {
            duration: 0,
            cycles : 0,
            dataAvailable : false,
            locationType: 0
        };
​
        try {
            // NodeID for InputArguments struct type (inherits from ScanSettings)
            const nodeID = new NodeId(NodeIdType.NUMERIC, 3010, 3);
            // Create ExtensionObject for InputArguments
            const scanSettingsObj = await session.constructExtensionObject(nodeID, scanSettingsParams);
            
            // Populate Method call with ExtensionObject as InputArgument
            const methodToCall = {
                    objectId: "ns=4;s=rfr310",
                    methodId: "ns=4;s=rfr310.ScanStart",
                    inputArguments: [
                        {
                            dataType: DataType.ExtensionObject,
                            value: scanSettingsObj
                        }
                    ]
                };
​
                // Call method, passing ScanSettings as input argument
                session.call(methodToCall,(err,results) => {
                    if (err) {
                        console.log(err);
                    } else {
                        console.log(results);
                    }
                });
​
        } catch (err) {
            console.log(err);
        }
    });
})();
`

HelloHungryImDad, 2020-04-29:
OK so now I need to understand the reverse problem. Having called the method, I'm now subscribing to events that it generates.

I am receiving the event data back, but the dataValue is (unsurprisingly) an OpaqueStructure:

```typescript
[
  Variant {
    dataType: 22,
    arrayType: 0,
    value: OpaqueStructure {
      nodeId: [NodeId],
      buffer: <Buffer 00 00 00 00 03 00 00 00 45 50 43 03 00 00 00 03 00 00 00 00 28 0a 00 00 00 32 38 30 30 30 39 31 38 36 00 00 00 00 00 f0 f9 e7 49 43 1e d6 01 07 00 00 ... 141 more bytes>    
    },
    dimensions: null
  }
]
```

 What is the function to construct an ExtensionObject from this? I've tried `promoteOpaqueStructure` but it is returning `undefined`.

erossignon, 2020-05-04:
promoteOpaqueStructure  takes a DataValue rather than a Variant as a argument.

You can convert the variant by calling promoteOpaqueStructure this way

````
promoteOpaqueStructure ({ value: myVariant });
````

HelloHungryImDad, 2020-05-05:
```javascript
            monitoredItem1.on("changed", async (dataValue) => {
                try{
                    console.log(dataValue);
                    console.log(await promoteOpaqueStructure({value: dataValue[0]}));
                } catch(e) {
                    console.log(e);
                }
            });
```
Produces 
```
TypeError: Cannot read property 'filter' of undefined
    at C:\Users\me1rjco\Documents\nodejs\node-opcua-test\node_modules\node-opcua\node_modules\node-opcua-client-dynamic-extension-object\dist\promote_opaque_structure.js:32:44
    at Generator.next (<anonymous>)
    at C:\Users\me1rjco\Documents\nodejs\node-opcua-test\node_modules\node-opcua\node_modules\node-opcua-client-dynamic-extension-object\dist\promote_opaque_structure.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (C:\Users\me1rjco\Documents\nodejs\node-opcua-test\node_modules\node-opcua\node_modules\node-opcua-client-dynamic-extension-object\dist\promote_opaque_structure.js:4:12)
    at promoteOpaqueStructure (C:\Users\me1rjco\Documents\nodejs\node-opcua-test\node_modules\node-opcua\node_modules\node-opcua-client-dynamic-extension-object\dist\promote_opaque_structure.js:30:12)
    at ClientMonitoredItemImpl.<anonymous> (C:\Users\me1rjco\Documents\nodejs\node-opcua-test\subscription-test.js:66:39)
    at ClientMonitoredItemImpl.emit (events.js:311:20)
    at ClientMonitoredItemImpl._notify_event (C:\Users\me1rjco\Documents\nodejs\node-opcua-test\node_modules\node-opcua\node_modules\node-opcua-client\dist\private\client_monitored_item_impl.js:145:18)
    at ClientSubscriptionImpl.__on_publish_response_EventNotificationList (C:\Users\me1rjco\Documents\nodejs\node-opcua-test\node_modules\node-opcua\node_modules\node-opcua-client\dist\private\client_subscription_impl.js:544:31)
```

Passing `dataValue.outputArguments[0]` produces `TypeError: Cannot read property '0' of undefined`

HelloHungryImDad, 2020-05-05:
The function declaration in v2.5.9 is

```
export declare function promoteOpaqueStructure(session: IBasicSession, dataValues: DataValue[]): Promise<void>;
```
If I call it in this way (i.e. `promoteOpaqueStructure(session, dataValue)`) I don't get the filter error, but I just get `undefined` result.

HelloHungryImDad, 2020-05-05:
The solution you posted does not work for me as it produces  a `dataValues.filter` is not a function.

Instead if I skip promoteOpaqueStructure and make my own call to resolveDynamicExtensionObject:
```javascript
const extraDataTypeManager = await getExtraDataTypeManager(session);
monitoredItem1.on("changed", async (dataValue) => {
                try{
                    await resolveDynamicExtensionObject(dataValue[0], extraDataTypeManager);
                    console.log(dataValue[0]);
                    
                } catch(e) {
                    console.log(e);
                }
            });
```

Where `dataValue[0]` is 
```javascript
  Variant {
    dataType: 22,
    arrayType: 0,
    value: OpaqueStructure {
      nodeId: [NodeId],
      buffer: <Buffer 00 00 00 00 03 00 00 00 45 50 43 03 00 00 00 03 00 00 00 00 28 0a 00 00 00 32 38 30 30 30 39 31 38 36 00 00 00 00 00 f0 f9 e7 49 43 1e d6 01 07 00 00 ... 141 more bytes>    
    },
    dimensions: null
  }
```

the value is successfully resolved to:
```javascript
Variant {
  dataType: 22,
  arrayType: 0,
  value: RfidScanResult {
    codeType: 'EPC',
    scanData: ScanData { switchField: 3, epc: [ScanDataEpc] },
    timestamp: 2020-05-05T09:09:28.567Z { high_low: [Array], picoseconds: 0 },
    location: undefined,
    sighting: [ [RfidSighting] ]
  },
  dimensions: null
}
```

HelloHungryImDad, 2020-05-05:
OK so I got this to work with the method you suggested:

```javascript
monitoredItem1.on("changed", async (dataValue) => {
                try{
                    const data = [{value: dataValue[0]}];
                    await promoteOpaqueStructure(session, data);
                    console.log(data[0].value.value);
                } catch(e) {
                    console.log(e);
                }
            });
```
However, I find it a bit odd that you have do do the dataValue conversion so instead I opted for the following (adapted from the content of the promoteOpaqueStructure function):

```javascript
const extraDataTypeManager = await getExtraDataTypeManager(session);
 monitoredItem1.on("changed", async (dataValue) => {
                try{
                    await resolveDynamicExtensionObject(dataValue[0], extraDataTypeManager);
                    console.log(data[0]);
                } catch(e) {
                    console.log(e);
                }
            });
```
Which to me seems more intuitive.

erossignon, 2020-05-05:
@HelloHungryImDad , you're right, my answer was not accurate enough, I have edited it.

erossignon, 2020-05-05:
@HelloHungryImDad ,for sure it could be more intutitive, I was providing a workaround so you can get sorted immediatly. With more time and more sponsors I will be able to accelerate the improvement of the remaining bits and make it more seemless ... I would be glad to have you as a sponsor !

teropes, 2020-05-06:
@HelloHungryImDad I had some struggle with the `promoteOpaqueStructure` as well until I noticed that it accepts `array` of dataValues. That's bit confusing since the subscription event provides singular dataValue so you have to wrap it in array. 

I have my own OpcUaClient class which provides me some easy of use wrappers for `read`, `write`, `subscribe` etc. and here is my working `subscribe` function:
```
  /**
   *  Register listener for data changes.
   */
  async subscribe(nodeId, dataChanged, options, indexRange) {
    const self = this;

    // fail if not ready
    if (!self.dataSubscription || !self.opcSession) return Error("data subscription or session not ready for subscription.");

    const defaultOptions = {
      samplingInterval: 100,
      discardOldest: true,
      queueSize: 1
    };

    let monitoredItemOptions = options || {};
    _.defaults(monitoredItemOptions, defaultOptions);
    
    const completeNodeId = (!nodeId.includes("ns=")) ? `ns=${self.ns};${nodeId}` : nodeId ;

    const itemToMonitor = {
      nodeId: completeNodeId,
      attributeId: opcua.AttributeIds.Value,
      indexRange: indexRange ? indexRange : null
    };

    const monitoredItem = await self.dataSubscription.monitor(itemToMonitor, monitoredItemOptions, opcua.TimestampsToReturn.Both);

    // bind provided dataCallback to data changes
    monitoredItem.on("changed", async function(dataValue) {
      await opcua.promoteOpaqueStructure(self.opcSession, [dataValue]);
      dataChanged(dataValue.value.toJSON().value, nodeId);
    });

  }
```

Notice that `await opcua.promoteOpaqueStructure(self.opcSession, [dataValue]);` it is wrapped in array. Underlying code is trying to use built-in `array.filter` functionality and that is why you get the error if you don't pass it as array.

HelloHungryImDad, 2020-05-06:
Thanks for your assistance @erossignon @teropes.

DavoodCH, 2020-05-10:
Dear developers,
Very interesting topic!
Following the examples above, it's not yet 100% clear for me how to implement structured OPC UA data access using extension objects in a custom client.
I think that a simple example and/or documentation could be very helpful for the community, for example as addition of the simple server example.
Additional questions:
- Are nested data types supported?
- Did I understand correctly that the data types must be defined directly in the Client?
Couldn't the data types be identified by recursive browsing of the server namespace?
Thanks for some additional information!
Davood

HelloHungryImDad, 2020-05-10:
@DavoodCH see my method call example above. To use an extension object on the client, all you need to know is the node ID of the structure's datatype on the server. The client then interrogates the server to unpack the structure into a js object. The object I create before calling constructExtensionObject is how you can initialize the extension object with defined values. It is optional and an empty object can be used to instantiate with default values.

As for nested structs, yes this appears to be the case. In my method response post above, the server is returning a variation of a [ScanResult](https://reference.opcfoundation.org/v104/AutoID/DataTypes/ScanResult/) structure, which contains nested structures and unions. These appear to be correctly (and recursively) unpacked.

DavoodCH, 2020-05-21:
@HelloHungryImDad Thank you very much for your explanations.
Your hints helped me to implement a Javascript OPC UA client using node-opcua, which is able to read out the tags and values of OPC UA extension objects, even if the structures are nested - Ok so far!
In order to give something back to the community I stripped down my code and created a simple yet fully functional client for reading extension objects. (see code below)
I hope this facilitates the task for future adopters.

Although the code is functional, I was still facing some **issues:**
- In a test bench, the solution is working fine. As soon as I try to read out extension objects from a OPC UA server (plc) that contains much more data/structures in the namespace, the client throws error messages and cannot decode the corresponding data type anymore. Is there some kind of library internal buffer that could potentially reach its maximum size when decoding extension objects from a server that provides high amount of data?

- It performance of the extension object reading/decoding looked rather poor.
Setting the sampling interval smaller than 1000mS leads to errors.
=> Are the contents of the extension objects decoded during every subscription "on changed" event?
=> Is it possible to decode the extension objects only once, for example when creating the subscription and then use the result as a template for future subscription events? This would probably save a lot of communication time and resources?

- Is it possible to read the extension objects using the OPC UA "registered read" principle, instead of subscriptions?

I would be very thankful for some hints and explanations.

Davood

```
// -- dependencies --//
const {
    OPCUAClient,
    AttributeIds,
    makeBrowsePath,
    ClientSubscription,
    TimestampsToReturn,
    ClientMonitoredItem,
    promoteOpaqueStructure,
} = require("node-opcua");

// -- configurations --//
var endpointUrl = "opc.tcp://192.168.0.180:4840";// PLC IP address.

// configure node id of variable to be monitored (extension object)
var nodeId = "ns=3;s=\"DBGlobal\".\"s_l1\"";

// -- application --//
const client = OPCUAClient.create();

async function main() {
    try {
        await client.connect(endpointUrl);
        console.log("connected !");

        const session = await client.createSession();
        console.log("session created !");

        // Creating subscription
        const subscription = ClientSubscription.create(session, {
            requestedPublishingInterval: 100, // publishing interval in ms. In PLC the "Current" variable increments every 100ms.
            requestedLifetimeCount: 100,
            requestedMaxKeepAliveCount: 10,
            maxNotificationsPerPublish: 100,
            publishingEnabled: true,
            priority: 10
        });

        subscription.on("started", function () {
            console.log("subscription started - subscriptionId=", subscription.subscriptionId);
        }).on("keepalive", function () {
            console.log("keepalive");
        }).on("terminated", function () {
            console.log("!!!The OPC UA Connection terminated!!!");
        });

        // monitoring parameters
        const parameters = {
            samplingInterval: 1000,// sampling interval for subscribed variables (monitored variables)
            discardOldest: true,
            queueSize: 10
        };

        // add monitored items to subscription
        var itemToMonitor1 = {
            nodeId: nodeId,
            attributeId: AttributeIds.Value
        }
            
        // add monitored item to subscription
        // save item to monitored items array
        const monitoredItem1 = ClientMonitoredItem.create(
            subscription,
            itemToMonitor1,
            parameters,
            TimestampsToReturn.Both)

        // wait for subscription event and evaluate structure
        monitoredItem1.on("changed", async (dataValue) => {
            try{
                //const data = [{value: dataValue[0]}];
                await promoteOpaqueStructure(session, [dataValue]);
                //console.log(dataValue);
                console.log(dataValue.value.toJSON().value)
            } catch(e) {
                console.log(e);
            }
        });

    } catch (err) {
        console.log("An error has occured : ", err);
    }
}

console.log("Starting OPC UA connection asynchronously");
main();
console.log("Asynch connection started.");
```

erossignon, 2020-05-30:
node-opcua@2.6.1 is now released 
@HelloHungryImDad: monitored ExtensionObject are now automatically decoded to Javascript object and are no longer opaqueStructure. please confirm.
@DavoodCH : decoding ExtensionObject  should  be quick and effective except for the first time an extension object is encourtered. In this case, node-opcua need to interogate the server to get the description of the exposed extension object. This should be a one off.
Also, a bug has been fixed that causes extension object being monitored to be constantly notified even when . Please confirm that you issue is fixed.

@DavoodCH 
> Are nested data types supported?
yes , Union and ExtensionObject with optional fields are also supported.

> Did I understand correctly that the data types must be defined directly in the Client?
No, node-opcua takes this in charge for you , there is no need to import any code, ExtensionObjects work straight out of the box. 
If you are using Typescript, though, you may have to declare a dedicated interface to expose the definition of your object.

HelloHungryImDad, 2020-06-03:
@erossignon Using 2.6.4 it doesn't seem to be automatically decoding OpaqueStructures.

```javascript
const { 
    OPCUAClient, 
    resolveNodeId, 
    AttributeIds, 
    TimestampsToReturn,
    constructEventFilter,
} = require("node-opcua");

const publishInterval = 1000;

const endpointUri = "opc.tcp://192.168.0.61:4840";

(async () => {

    const client = OPCUAClient.create({ endpoint_must_exist: false});
    client.on("backoff", () => console.log("Backoff: trying to connect to ", endpointUri));

    await client.withSessionAsync(endpointUri, async (session) => {

        try {

            // Create subscription with options
            const subscription = await session.createSubscription2({
                maxNotificationsPerPublish: 1000,
                publishingEnabled: true,
                requestedLifetimeCount: 100,
                requestedMaxKeepAliveCount: 10,
                requestedPublishingInterval: publishInterval
            });

            
            // HaRfidScanResult event type has address ns=4;i=1006
            // and include the fields below
            var fields = [
                "3:ScanResult",
            ];
            
            // Create event filter for when changes are detected on the server
            const eventFilter = constructEventFilter(fields);
            const monitoringParameters = {
                discardOldest: true,
                queueSize: 100,
                samplingInterval: 0, // when ever changed
                filter: eventFilter
            };
            
            // Monitor the event notifier. This could be any tag you wish
            // However in this case new RFID tag events are generated 
            // by the event notifier
            const itemToMontior1 = {
                nodeId: resolveNodeId("ns=0;i=2253"),    // Server
                attributeId: AttributeIds.EventNotifier
            };
            const monitoredItem1 = await subscription.monitor(
                itemToMontior1,
                monitoringParameters,
                TimestampsToReturn.Both,
            );

            // Callback function for when an event is generated
            monitoredItem1.on("changed", async (dataValue) => {
                try {
                    console.log(dataValue);

                } catch(e) {
                    console.log(e);
                }
            });

            await new Promise(resolve => setInterval(() => {
            }, 1000));

        } catch (err) {
            console.log(err);
        }
    });
})();
````

Produces:

```
[
  Variant {
    dataType: 22,
    arrayType: 0,
    value: OpaqueStructure {
      nodeId: [NodeId],
      buffer: <Buffer 00 00 00 00 03 00 00 00 45 50 43 03 00 00 00 03 00 00 00 00 30 0c 00 00 00 30 39 ec bc 01 b5 b0 41 2a 05 f7 d9 00 00 00 00 50 34 08 6c 8c 39 d6 01 01 ... 23 more bytes>
    },
    dimensions: null
  }
]
```

In UAExpert, the contents should be [this structure](https://imgur.com/a/bz4KCMy).

erossignon, 2020-06-03:
@HelloHungryImDad  well spotted, one case was missing to automatically promote opaque structure to json extension object  in EventNotificationList.... This will be fixed in #798

erossignon, 2020-06-03:
@HelloHungryImDad 

in your example, you are listening to Event rather that Notification on UAVariable.

the `on("changed",()=>{});` lambda has a different signature in this case 

```
  monitoredItem1.on("changed", async (eventFilelds: Variant[]) => {
  };
```
rather than 
```
  monitoredItem1.on("changed", async (dataValue: DataValue) => {
     
  };
```

GeneL, 2020-06-08:
Hi All,
Looking in the code and documentation I could not find a TypeScript code example to create a user defined structured extension object on the server side. Could you please point to an example or provide a code sample?

DavoodCH, 2020-06-18:
Thank you very much for your hints!

I measured the read times, the communication times are pretty decent when reading extension objects.
Reading a structure containing about 6000 tags and large amount of nested types we got the following results:
=> ~250mS for the first run when the extension objects are requested from the server
=> ~20mS for a large structure containing about 6000 tags and lots of nested types

@erossignon:
As soon as we load additional data to the plc, the client is no more able to read/decode the data altough we are reading still the same structure and the additional data is not read by the client.

error message:
-----------
Error: Cannot find getDataTypeAndEncodingId for Rep_BMD_ProcStepCtrl
    at _getOrCreateStructuredTypeSchema (/home/edge/Documents/node_modules/node-opcua-schemas/dist/source/tools.js:123:19)
    at Object.getOrCreateStructuredTypeSchema (/home/edge/Documents/node_modules/node-opcua-schemas/dist/source/tools.js:138:12)
    at /home/edge/Documents/node_modules/node-opcua-schemas/dist/source/parse_binary_xsd.js:300:21
    at SaxLtx.<anonymous> (/home/edge/Documents/node_modules/node-opcua-xml2json/dist/source/xml2json.js:323:17)
    at SaxLtx.emit (events.js:315:20)
    at SaxLtx.<anonymous> (/home/edge/Documents/node_modules/node-opcua-xml2json/dist/source/xml2json.js:309:24)
    at SaxLtx.emit (events.js:315:20)
    at SaxLtx._handleTagOpening (/home/edge/Documents/node_modules/ltx/lib/parsers/ltx.js:40:12)
    at SaxLtx.write (/home/edge/Documents/node_modules/ltx/lib/parsers/ltx.js:159:18)
    at Xml2Json.parseString (/home/edge/Documents/node_modules/node-opcua-xml2json/dist/source/xml2json.js:230:16)
-----------
any ideas?

Thank you very much for your help!

erossignon, 2020-06-18:
@DavoodCH  this would require deeper investigation, can you make the server available remotely ?

erossignon, 2020-06-20:
@DavoodCH , thank you for providing access to the Siemens plc device. 
The issue is fixed in [node-opcua@2.7.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.7.0)

seang96, 2020-06-22:
When trying to write to an Extension Object node I am getting "A variant with DataType.ExtensionObject must have a ExtensionObject value". Is there a way to convert the value to the ExtensionObject type?

HelloHungryImDad, 2020-06-23:
@seang96 See my snippet above from 29th April for constructing extension objects. Your error suggests the method you are using is not properly creating an ExtensionObject.

seang96, 2020-06-23:
@HelloHungryImDad, your example doesn't work for my need since I need a
generic way of getting the DataType of the Extension Object. I edited my
comment and achieved that but in doing so I have an additional read of the
node required to do a write which seems a little absurd to do for
performance purposes.

erossignon, 2020-06-23:
@seang96, you definitely need to know the dataType of the object you need to write, so you can instantiate it.*
You could simplify a bit this way:

```javascript
const nodeId = "ns=1;s=MyExtensionObjectVariable";
const dataTypeNodeId = await session.read({
    nodeId, 
    attributeId: AttributeIds.DataType
}).value.value as NodeId;
 
const extensionObj= session.constructExtensionObject(dataTypeNodeId ,{
 /* pojo*/
});
const variant = new Variant({
   dataType: DataType.ExtensionObject,
   value: extensionObj,
 });

 const statusCode = await session.write({
      nodeId,
      attributeId: AttributeIds.Value,
      value: /*DataValue*/ { 
         value: variant,
         statusCode: StatusCodes.Good
      }
});
```

mrpfunk, 2020-06-24:
I have to write a server offering a Extension Object.
How is that done? What needs to be set to the dataType field?
`
const eo = {x: 1, y: 2 }
namespace.addVariable({
    componentOf: device,
    nodeId: "s=position1",
    browseName: "Position1",
    dataType: opcua.ExtensionObject,  // ????
    value: {
        get: function () {
            return new opcua.ExtensionObject({dataType: opcua.DataType.ExtensionObject, value: eo })
        }
    }
})
`

erossignon, 2020-06-24:
@mrpfunk an fully documented example of a server exposing a ExtensionObject can be found inside [NodeOPCUA by Example](https://leanpub.com/node-opcuabyexample) , chapter 11.

seang96, 2020-06-25:
@erossignon, I am testing on how the OPC-UA Client handles an ExtensionObject when the OPC-UA server does changes. I also have restarted OPC-UA Client and these results are still the same. Shut off the device hosting the server and turned it back on, results are still the same. Shutting down the device and the client, starting client then the device and the results are still the same. I figured they wouldn't update if the DataType has been looked at previously, but I didn't expect to permanently no longer be able to read them. I can read the variables in the structure but not the entire structure itself. UAExpert updates fine after a rebrowse.

1. Add new variable that is a new structure, OPC-UA client fails to read the value resulting in an empty object {} for value.
2. Change existing variable to a different existing structure. Value is {}.

erossignon, 2020-06-25:
@seang96, we probably need to review your implementation code. Can you share it ? or contact me in private ?

IagoSRL, 2020-10-15:
> 
> 
> @mrpfunk, a fully documented example of a server exposing an ExtensionObject can be found inside [NodeOPCUA by Example](https://leanpub.com/node-opcuabyexample) , chapter 11.

Hi @erossignon, I'm interesting on learning more about managing arrays of ExtensionObjects, to read/write at the client, and emulating them at a debugging server with node-opcua (to run tests without the real PLC). The "Table of Contents" of the book at leanpub doesn't contains chapter 11: Does it mean that it is not published yet? Will I found at the current state of the book the documentation needed for that requirements?

Thanks in advance!

erossignon, 2020-10-15:
@IagoSRL , good catch : The TOC is out of date .. I have now updated it.

closed 2020-10-15 by erossignon