node-opcua/node-opcua #321: Notification (server publish) example

enobufs, 2017-03-10:
I would like to learn how to write server code to send a notification (publish) to a client. Is there a sample code for it? (I am aware node-opcua-sample has a client-side example.)

erossignon, 2017-03-10:
Subscriptions/MonitoredItem is a built-in mecanism of OPCUA. You don't need to do extract work to make your variable publishable.
A sample server can be found at  
 - https://github.com/node-opcua/node-opcua-sampleserver/blob/master/server.js

enobufs, 2017-03-11:
Ah, I see. Now I get it. I am looking at the simple_client.js in the node-opcua-sample as well, and it is requesting the server to publish every 1000 msec.

I believe there would be a case where a client want's to receive a notification only when a variable on the server was changed. Would this be possible? If so, I would like to see a sample code if possible.

erossignon, 2017-03-11:
Yes, this behaviour is already built-in the OPC-UA. There is no extra work.
publishInterval is just the maximum rate the server will communicate monitored value changed back to the client. If the monitored value doesn't change you will not received any value change notification. A keep-alive mechanism will send a empty notification change from time to time so the connection doesn't drop in case all monitored variables are still.
For greater details please refer to the OPCUA specification part 4 - 5.12 MonitoredItem Service Set. OPC-UA specs can be downloaded from https://opcfoundation.org/

enobufs, 2017-03-11:
Thanks for referring me to the spec. Now I understand how the notification works with this module!

closed 2017-03-11 by enobufs