node-opcua/node-opcua #322: How to write NULL to a node value with writeValue from client side

tomsoftware, 2017-03-14:
Sometimes my sensor can't read a value and returns NULL. Because of my infrastructure I need to update the node-value of my Node-OPC-UA Server via a OPC UA Client but I don't know how to set it "null". I get result: "BadTypeMismatch" if I set the value:=NULL and use writeValue.
At the moment I use a StatusCodes:=BadDataUnavailable and a value:=0 but don't feel happy with that.

Thank you.
Regards Thomas

erossignon, 2017-10-07:
How about this:

From within a server:
```javascript
myvariable.setValueFromSource({dataType: DataType.Null},StatusCodes.BadDataUnavailable);
```
Using a client
```javascript
           var nodesToWrite = [
                {
                    nodeId: nodeId,
                    attributeId: opcua.AttributeIds.Value,
                    value: /*new DataValue(*/{
                        statusCode: opcua.StatusCodes.BadDataUnavailable,
                        value: {/* Variant */dataType: opcua.DataType.Null}
                    }
            
                }
            ];
            session.write(nodesToWrite, function (err) {
                callback(err);
            });
```

closed 2017-10-07 by erossignon

tomsoftware, 2017-10-12:
Thank you, that fixes the issue!