node-opcua/node-opcua #324: Events on address space change

ascatox, 2017-03-23:
Hi All,
I'm encountering a big problem with the sdk.
I've an address space changing over the time, new **Objects** could be added to the address space.
There's an api **to notify** these changes to the clients, in order to subscribe/monitor these new Objects?
Thanks in advance.

erossignon, 2017-03-23:
Yes, the OPCUA specification recommend that changes in the address space on the server side are notified to the clients using a dedicated Event that can be monitored on the "Server" (rootFolder.Objects.Server) node in the addressSpace. (The server on which you are trying to connect shall have this feature in place, not all servers implement this)

ascatox, 2017-03-23:
Is it possible implement this feature server side with node-opcua SDK?

erossignon, 2017-03-31:
Absolutely, this is implemented as per specification. 
You can refer to the corresponding [unit test](https://github.com/node-opcua/node-opcua/blob/68b1b57dec23a45148468fbea89ab71a39f9042f/test/address_space/test_address_space_model_change_event.js#L11) to understand how it works.

closed 2017-09-18 by erossignon