node-opcua/node-opcua #330: EventType_Schema like DataType_Schema

biancode, 2017-03-28:
Hi,

it would be great to have an EventType_Schema like the DataType_Schema.

erossignon, 2017-03-28:
The trouble is that DataType is a  well defined  enumeration with keys and values defined in the specification, but EventType is a nodeId that can refer to a EventType node defined in the standard UA namespace or in any custom namespace ... therefore it is difficult to prebuild.

Please provide an description of what you're trying to achieve ... do you care about this inside a client application or a server application ?

biancode, 2017-03-29:
Dear Etienne,
it is for the client just to have the basic EventTypes for selection to start with events in OPC UA. For me, it would be great to have an enum of code_gen/NodeIds.csv -> '*EventTyp,' to provide a simple select field. Or is there a way to search for all possible events of a namespace, eventarea or for a root Node-Id and the child nodes to provide it in a select. (key: OPCEventAreaBrowser?)

erossignon, 2017-03-29:
What about this  ? 
https://gist.github.com/erossignon/df6ffd19ca0ff228395fc95a688310ac

erossignon, 2017-04-06:
@biancode  how this helped ?

biancode, 2017-04-07:
I'll work on this weekend - have to find some time ;)

biancode, 2017-04-08:
Perfect! Works very well! Thank you very much! 

That is my source now:

```
de.biancoroyal.opcua.iiot.core.listener.getAllEventTypes = function (session, callback) {
  let entries = []
  let makeNodeId = this.core.nodeOPCUA.makeNodeId
  let ObjectTypeIds = this.core.nodeOPCUA.ObjectTypeIds

  let browseEventTypes = {
    nodeId: makeNodeId(ObjectTypeIds.BaseEventType),
    referenceTypeId: this.core.nodeOPCUA.resolveNodeId('HasSubtype'),
    browseDirection: this.core.nodeOPCUA.browse_service.BrowseDirection.Forward,
    includeSubtypes: true,
    nodeClassMask: this.core.nodeOPCUA.browse_service.NodeClassMask.ObjectType,
    resultMask: 63 // all from ResultMask_Schema - a const with name ALL would be great
  }

  let nodesToBrowse = [browseEventTypes]

  session.browse(nodesToBrowse, function (err, results, diagnostics) {
    if (err) { callback(err) }
    results[0].references.forEach(function (reference) {
      entries.push({displayName: reference.displayName.text, nodeId: reference.nodeId, reference: reference})
    })

    callback(null, entries, diagnostics)
  })
}
```

biancode, 2017-04-08:
opcuaIIoT:listener Listener Start OPC UA Session +0ms
  opcuaIIoT:listener AuditEventType : ns=0;i=2052 +213ms
  opcuaIIoT:listener SystemEventType : ns=0;i=2130 +1ms
  opcuaIIoT:listener BaseModelChangeEventType : ns=0;i=2132 +0ms
  opcuaIIoT:listener TransitionEventType : ns=0;i=2311 +0ms
  opcuaIIoT:listener ConditionType : ns=0;i=2782 +0ms
  opcuaIIoT:listener EventQueueOverflowEventType : ns=0;i=3035 +0ms
  opcuaIIoT:listener ProgressEventType : ns=0;i=11436 +0ms
  opcuaIIoT:listener GenerateValuesEventType : ns=2;i=9371 +1ms

closed 2017-04-08 by biancode