node-opcua/node-opcua #332: Read / write access to variables on a UA proxy service

enobufs, 2017-03-30:
When I was writing a proxy service (see below), I stumbled upon a difficulty.

```
 Device          Proxy              Application
            +--------------+       +--------+
  +-+       |+-+        +-+|       |+-+     |
  |U|       ||U| bridge |U||       ||U|     |
  |A|<------>|A|<------>|A|<------->|A| App |
  |S|       ||C|        |S||       ||C|     |
  |0|       ||0|        |1||       ||1|     |
  +-+       |+-+        +-+|       |+-+     |
            +--------------+       +--------+
```

I wanted to write a proxy server with these reasons:
* To overcome load on a computationally constrained end devices.
* Easier to achieve access controls
* To enable history feature for devices that do not support it. (see #331)
* Conceptually allowed by OPC UA spec

In this configuration, UAS0 has a variable X and application wants to read, you would also define the corresponding variable in UAS1, using AddressSpace.addVariable().

```js
addressSpace.addVariable({
    componentOf: device01Node,
    nodeId: "ns=1;s=X",
    browseName: "X-axis",
    dataType: "Double",
    value: {
        get: () => {
            return new opcua.Variant({dataType: opcua.DataType.Double, value: this._x });
        }
    }
});
```

The actual variable 'this._x' is fetched with the get method but it is `synchronous` call. Consequently, UAS1 has no chance to call read() on UAC0 to retrieve the current X value on UAS0.

Ideally the `get()` should be an `async` call, or allow another method like `getAsync(cb)` so that the proxying variables is possible.

One workaround is to implement this by having UAS1 monitor value X on UAS0 to keep this._x on UAS1 updated always, but this will incur unnecessary traffic on the network and CPU load on the system.

marcelwinh, 2017-03-30:
Hi,
i think that's exactly what you are looking for:

[http://node-opcua.github.io/api_doc/classes/UAVariable.html#method_bindVariable](http://node-opcua.github.io/api_doc/classes/UAVariable.html#method_bindVariable)

**Variation 3:**

> This variation can be used when the value associated with the variables requires a asynchronous function call to be extracted. In this case, the user should provide an async method refreshFunc.
> 
> The refreshFunc shall do whatever is necessary to fetch the most up to date version of the variable value, and call the callback function when the data is ready.
> 
> The callback function follow the standard callback function signature:
> 
> the first argument shall be null or Error, depending of the outcome of the fetch operation,
> the second argument shall be a DataValue with the new UAVariable Value, a StatusCode, and time stamps.
> Optionally, it is possible to pass a sourceTimestamp and a sourcePicoseconds value as a third and fourth arguments of the callback. When sourceTimestamp and sourcePicoseconds are missing, the system will set their default value to the current time..

```javascript
...
var options =  {
   refreshFunc : function(callback) {
     ... do_some_async_stuff_to_get_the_new_variable_value
     var dataValue = new DataValue({
         value: new Variant({...}),
         statusCode: StatusCodes.Good,
         sourceTimestamp: new Date()
     });
     callback(null,dataValue);
   }
};
...
engine.bindVariable(nodeId,options):
...
```

enobufs, 2017-03-30:
FANTASTIC! IT WORKS! Thanks @marcelwinh 😄

enobufs, 2017-03-30:
Sorry,  one more thing...  Is there an async setter?
Looking at the bindVariable code, I guess it is to do...

erossignon, 2017-03-30:
You'll need to provide a _timestamped_set_func to your UAVariable.

enobufs, 2017-03-31:
I tried something like this:
```
addressSpace.addVariable({
    componentOf: device01Node,
    nodeId: "ns=1;s=abc",
    browseName: "ABC",
    dataType: "Double",

    value: {
        refreshFunc : (callback) => {
            :
        },
        timestamped_get() {
            :
        },
        timestamped_set(dataValue, callback) {
            :
        },
})
```

When added timestaped_set(), it complained about missing timestamped_get().
Then I added timestamped_get(), then it complained about the use of refreshFunc()... so I am stuck. :(

Could you tell me once again how I can make both setter and getter on the same variable async?

closed 2017-04-08 by erossignon