node-opcua/node-opcua #333: "readValueAsync has failed" in endless loop on Server if client crashs

tomsoftware, 2017-04-05:
If my UA Client crashs then in many cases the node-opcua Server falls in a endless loop with the error:
(calling this 10 times a second)

    func readValueAsync has failed
      stack AssertionError: _.isFinite(value)
        at coerceVariantType (\node_modules\node-opcua\lib\datamodel\variant_tools.js:63:13)
        at Object.construct_hook (\node_modules\node-opcua\schemas\Variant_schema.js:404:29)
        at new Variant (\node_modules\node-opcua\_generated_\_auto_generated_Variant.js:48:22)
        at UAVariable.timestamped_get [as _timestamped_get_func] (\node_modules\node-opcua\lib\address_space\ua_variable.js:1362:41)
        at UAVariable.async_refresh_func [as refreshFunc] (\node_modules\node-opcua\lib\address_space\ua_variable.js:916:30)
        at UAVariable.self.asyncRefresh (\node_modules\node-opcua\lib\address_space\ua_variable.js:874:26)
        at UAVariable.readValueAsync (\node_modules\node-opcua\lib\address_space\ua_variable.js:1272:14)
        at monitoredItem_read_and_record_value_async (\node_modules\node-opcua\lib\server\opcua_server.js:1813:10)
        at MonitoredItem.self [as samplingFunc] (\node_modules\node-opcua\lib\server\opcua_server.js:1854:13)
        at MonitoredItem._on_sampling_timer (\node_modules\node-opcua\lib\server\monitored_item.js:428:27)
      SAMPLING ERROR => { AssertionError: _.isFinite(value)
        at coerceVariantType (\node_modules\node-opcua\lib\datamodel\variant_tools.js:63:13)
        at Object.construct_hook (\node_modules\node-opcua\schemas\Variant_schema.js:404:29)
        at new Variant (\node_modules\node-opcua\_generated_\_auto_generated_Variant.js:48:22)
        at UAVariable.timestamped_get [as _timestamped_get_func] (\node_modules\node-opcua\lib\address_space\ua_variable.js:1362:41)
        at UAVariable.async_refresh_func [as refreshFunc] (\node_modules\node-opcua\lib\address_space\ua_variable.js:916:30)
        at UAVariable.self.asyncRefresh (\node_modules\node-opcua\lib\address_space\ua_variable.js:874:26)
        at UAVariable.readValueAsync (\node_modules\node-opcua\lib\address_space\ua_variable.js:1272:14)
        at monitoredItem_read_and_record_value_async (\node_modules\node-opcua\lib\server\opcua_server.js:1813:10)
        at MonitoredItem.self [as samplingFunc] (\node_modules\node-opcua\lib\server\opcua_server.js:1854:13)
        at MonitoredItem._on_sampling_timer (\node_modules\node-opcua\lib\server\monitored_item.js:428:27)
        name: 'AssertionError',
        actual: undefined,
        expected: undefined,
        operator: undefined,
        message: '_.isFinite(value)',
        generatedMessage: false }

I'm trying to find more information about this bug but may you have a idea?

Thank you.

erossignon, 2017-04-06:
Can you provide the implement of your UAVariable value getter and setter for inspection ?

tomsoftware, 2017-04-10:
Hello Erossignon,
I don't use any self made getter or setter. I load the model from xml, create one instace add some historical "node-properties" and read and write variables from the Server with my UA client.

erossignon, 2017-04-10:
OK , can you provide the xml file then, or at least a bare minimum xml file that would allow us to reproduce the behavior.

tomsoftware, 2017-04-13:
Thank you! ... This is difficult for me because it only occurs sporadically. I will try next week to create an small "example".
I use the OPCFoundation/UA-.NET to write and read values from the node-Ua-Server.

erossignon, 2018-01-31:
@tomsoftware Thomas, is this still an issue with the most recent version of node-opcua ( ie 0.2.1 ?)
Let me know.

tomsoftware, 2018-06-01:
this issue is no longer appearing on may system. Don't know what it caused. Thanks for you help.

closed 2018-06-01 by tomsoftware