node-opcua/node-opcua #334: Removing a monitored item

gcperrin, 2017-04-06:
It looks like the implementation is generally unfinished around removing a monitored item. A patch including a public wrapper (and coorsponding test) around the following function, from _lib/client/client_subscription.js_ would be complete for a passing pull request? Or is there a function hiding elsewhere that provides the functionality.

@erossignon If you confirm I'll take a crack at it.

Best,
G

```javascript
ClientSubscription.prototype._delete_monitored_item = function (monitoredItem, callback) {
    var self = this;
    assert(self.isActive());

    var clientHandle = monitoredItem.monitoringParameters.clientHandle;

    assert(self.monitoredItems[clientHandle], "expecting existing monitoredItem");

    self.session.deleteMonitoredItems({
        subscriptionId: self.subscriptionId,
        monitoredItemIds: [monitoredItem.monitoredItemId]
    }, function (err) {

        // remove monitored items
        self._remove(monitoredItem);
        assert(!self.monitoredItems[clientHandle], "monitored item should have been removed");
        callback(err);
    });
};
```

erossignon, 2017-04-06:
``` javascript
monitoredItem.terminate(function done() {
});
```
shall terminate and remove a single monitored item

gcperrin, 2017-04-06:
Thank you for the insight, much obliged. Thought it would be there due to monitors existence should have dug deeper! Can close.

erossignon, 2017-04-06:
you're welcome!

closed 2017-04-06 by erossignon