node-opcua/node-opcua #336: OpcUAClient doesn't seem to take securityPolicy and securityMode into account

Faerylnahr, 2017-04-12:
Hi,

first of all, thanks so much for your work and effort! It's a great project :-)

I was working with node-opcua by using node-red-contrib-opcua. When creating an endpoint I had some difficulties connecting to an OPC UA server which uses SecurityPolicy and SecurityMode others than None. I tried to track down the issue and came up with this line
`new opcua.OPCUAClient();`

I found out, that the opcua_client takes options in order to configure, e.g. endpoint_must_exist. But I didn't found any code snippet coping with 
>  * @param [options.securityMode=MessageSecurityMode.None] {MessageSecurityMode} the default security mode.
>  * @param [options.securityPolicy =SecurityPolicy.NONE] {SecurityPolicy} the security mode.

Where can I tell the client to use a certain securityPolicy or SecurityMode?
Any help appreciated :-)

erossignon, 2017-04-12:
This is a [example to use securityMode in a client](https://github.com/node-opcua/node-opcua/blob/c060e78b90e1ea139ec05373ec816891403b97ad/test/secure_connection/test_e2e_secure_UserNameIdentityToken.js#L80)

Faerylnahr, 2017-04-13:
Hi, thanks for the example :-) Looks like I've been on the right way 👍

closed 2017-04-13 by erossignon