node-opcua/node-opcua #337: Usage of valueRank and arrayDimensions

enobufs, 2017-04-12:
This is a question.

I would like to create a Variable that holds a 3-dimension value, such as a position (x, y, z), but I don't know if I am doing right.

Here's the code on the server:
```js
        addressSpace.addVariable({
            componentOf: device,
            nodeId: "ns=1;s=Position",
            browseName: "position",
            dataType: "Double",
            valueRank: 1,
            arrayDimensions: [3],
            value: {
                get: () => {
                    return new opcua.Variant({
                        dataType: opcua.DataType.Double,
                        value: [1, 2, 3]
                    });
                }
            }
        });
```

Then, I am reading the variable with `read()` method like this:
```js
                var max_age = 0;
                var nodes_to_read = [
                    { nodeId: "ns=1;s=Position", attributeId: opcua.AttributeIds.Value }
                ];
                this._session.read(nodes_to_read, max_age, (err, nodes_to_read, dataValues) => {
                    if (err) {
                        return done(err);
                    }
                    console.log("dataValues:" , JSON.stringify(dataValues, null, 4));
                    done();
                });
```

Which prints out:
```
dataValues: [
    {
        "value": {
            "dataType": "Double",
            "arrayType": "Array",
            "value": {
                "0": 1,
                "1": 2,
                "2": 3
            }
        },
        "statusCode": {
            "value": 0,
            "description": "No Error",
            "name": "Good"
        },
        "sourceTimestamp": "2017-04-12T22:10:13.829Z",
        "sourcePicoseconds": 0,
        "serverTimestamp": "2017-04-12T22:10:13.831Z",
        "serverPicoseconds": 0
    }
]
```

The returned `value`, with util.inspect() shows this:
```
       value:
        Float64Array [
          1,
          2,
          3,
          [BYTES_PER_ELEMENT]: 8,
          [length]: 3,
          [byteLength]: 24,
          [byteOffset]: 0,
          [buffer]: ArrayBuffer { byteLength: 24 } ],
       dimensions: null },
```

Is this working? (why the `dimensions` is null?)

erossignon, 2017-04-12:
A few tips:
You coud take benefit of the overloaded toString method to get a nice output
(this works on node-opcua objects) 
```javascript
  this._session.read(nodes_to_read, max_age, (err, nodes_to_read, dataValues) => {
  if (err) {
    return done(err);
  }
// console.log("dataValues:" , JSON.stringify(dataValues, null, 4));
  console.log(" dataValue:" , dataValues[0].toString());
  done();
});
```
Internally, Array of Floats, UInts etc are stored as Javascript typed Arrays  for compactness, hence Float64Array 

When you create a Variant containing a array of Float, for example, you can pass a standard Javascript array with floats number into it  (this is what you're doing here  : 
```javascript
 return new opcua.Variant({
 dataType: opcua.DataType.Double,
 value: [1, 2, 3] // <----- probably I am not doing right here... <<< YES you are doing OK here
});
```
node-opcua will build a Float64Array  for you.

```javascript
 return new opcua.Variant({
 dataType: opcua.DataType.Double,
 value: new Float64Array([1, 2, 3]) // This is also a valid way of creating a Variant as a array of Double
});
```

erossignon, 2017-04-12:
there are 3 types of Variant:
 Scalar, Array and Matrix,
dimensions is used when you want to build a Matrix

for instance a 3x3 matrix could be created this way:

```` javascript
var a = new opcua.Variant({
  dataType: opcua.DataType.Double, 
  arrayType: opcua.VariantArrayType.Matrix,
  dimensions: [3,3] , 
  value: [1,2,3,4,5,6,7,8,9]
});
```

erossignon, 2017-04-12:
This being said, at the OPCUAVaraible node level , valueRank and arrayDimensions are just a indication for the client of what the Value of the variable should be (informative only);
```
 valueRank: 1,
 arrayDimensions: [3],
```
I recommand that you refer to ```OPCUA 1.03 Specification Part 5 - 3.3 Convention for node description``` for  a complete explaination.

enobufs, 2017-04-12:
I see. Good to know that `value` only takes a single dimension array (not a nested array).

I have just tried the 3x3 matix like this (identical to yours):
```js
        addressSpace.addVariable({
            componentOf: device,
            nodeId: "ns=1;s=Position",
            browseName: "position",
            dataType: "Double",
            valueRank: 2,
            arrayDimensions: [3, 3],
            value: {
                get: () => {
                    return new opcua.Variant({
                        dataType: opcua.DataType.Double,
                        arrayType: opcua.VariantArrayType.Matrix,
                        dimensions: [3, 3],
                        value: [1, 2, 3, 4, 5, 6, 7, 8, 9]
                    });
                }
            }
        });
```

Then I got this error during the `read` call:
```
 Internal error reading  NodeId        ns=1;s=Position
                         AttributeId   13
                         Cannot read property 'reduce' of undefined
                         TypeError: Cannot read property 'reduce' of undefined
    at calculate_product (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/schemas/Variant_schema.js:25:17)
    at Object.construct_hook (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/schemas/Variant_schema.js:396:46)
    at new Variant (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/_generated_/_auto_generated_Variant.js:48:22)
    at new DataValue (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/_generated_/_auto_generated_DataValue.js:58:36)
    at _clone_with_array_replacement (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/lib/datamodel/datavalue.js:102:12)
    at extractRange (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/lib/datamodel/datavalue.js:132:21)
    at UAVariable.readValue (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/lib/address_space/ua_variable.js:411:21)
    at UAVariable.readAttribute (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/lib/address_space/ua_variable.js:570:25)
    at ServerEngine._readSingleNode (/Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/lib/server/server_engine.js:922:33)
    at /Users/ytakeda/Projects/enobufs/node-opcua-examples/node_modules/node-opcua/lib/server/server_engine.js:985:21
dataValues: [ DataValue {
    value: null,
    statusCode:
     { [Number: 2147614720]
       value: 2147614720,
       description: 'An internal error occurred as a result of a programming or configuration error.',
       name: 'BadInternalError' },
    sourceTimestamp: null,
    sourcePicoseconds: 0,
    serverTimestamp: null,
    serverPicoseconds: 0 },
  [length]: 1 ]
```

A bug or I am not doing right?

erossignon, 2017-04-13:
This is a bug , thank you

closed 2017-10-12 by erossignon