node-opcua/node-opcua #340: Can not read with attributeId

biancode, 2017-04-13:
Dear Etienne,

I get an error on simple read and I can not find the problem, because it looks all correct.

AttributeId's are 1, 3, and so on ...
item is a string like ns=1;s=TestReadWrite
maxAge is 0 on test

ERROR: TypeError: Cannot read property 'namespaceIndex' of undefined +0ms

```
for (item of itemsToRead) {
  transformedItem = {
    nodeId: item,
    attributeId: Number(node.attributeId)
  }
  transformedItemsToRead.push(transformedItem)
}
session.read(transformedItemsToRead, node.maxAge).then(function (resultsConverted, results, diagnostics) { ...
```

Please, can you give some help?

biancode, 2017-04-13:
nodesToRead was missing in the callback - solved / found

biancode, 2017-05-19:
Dear Etienne,

the maxAge is maybe not consistent on subscribing for events.
There is no read from client side, just ClientSubscription for events - server and client are node-opcua based

    //// options should have
    //var allowedProperties = [
    //    'requestedPublishingInterval',
    //    'requestedLifetimeCount',
    //    'requestedMaxKeepAliveCount',
    //    'maxNotificationsPerPublish',
    //    'publishingEnabled',
    //    'priority'
    //];

```
readRequest.maxAge >= 0
AssertionError: readRequest.maxAge >= 0
    at ServerEngine.read (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/server_engine.js:973:5)
    at /Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/opcua_server.js:1638:37
    at ServerEngine.refreshValues (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/server_engine.js:1543:16)
    at /Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/opcua_server.js:1634:23
    at OPCUAServer._apply_on_SessionObject (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/opcua_server.js:1327:5)
    at OPCUAServer._on_ReadRequest (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/opcua_server.js:1600:10)
    at OPCUAServer.OPCUABaseServer.on_request (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/base_server.js:230:21)
    at OPCUAServerEndPoint.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/opcua_server.js:272:14)
    at emitThree (events.js:116:13)
    at OPCUAServerEndPoint.emit (events.js:194:7)
    at ServerSecureChannelLayer.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/server_end_point.js:269:18)
    at emitOne (events.js:96:13)
    at ServerSecureChannelLayer.emit (events.js:188:7)
    at ServerSecureChannelLayer._on_common_message (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/server_secure_channel_layer.js:975:18)
    at emitMany (events.js:127:13)
    at MessageBuilder.emit (events.js:201:7)
    at MessageBuilder._decode_message_body (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/misc/message_builder.js:445:22)
    at MessageBuilderBase._feed_messageChunk (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/misc/message_builder_base.js:189:18)
    at PacketAssembler.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/misc/message_builder_base.js:49:14)
    at emitOne (events.js:96:13)
    at PacketAssembler.emit (events.js:188:7)
    at PacketAssembler.feed (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/transport/packet_assembler.js:102:14)
    at MessageBuilderBase.feed (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/misc/message_builder_base.js:153:30)
    at ServerTCP_transport.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/server/server_secure_channel_layer.js:339:37)
    at emitOne (events.js:96:13)
    at ServerTCP_transport.emit (events.js:188:7)
    at ServerTCP_transport._on_message_received (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/transport/tcp_transport.js:186:14)
    at PacketAssembler.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/transport/tcp_transport.js:283:30)] (l=1)
```

It should be 0 if it is not set, shouldn't it?

Thank you!
Klaus

biancode, 2017-05-19:
```
 */
ClientSession.prototype.read = function (nodesToRead, maxAge, callback) {

    var self = this;

    if (!callback) {
        callback = maxAge;
        maxAge = 0;
    }

	maxAge = maxAge || 0;
```

maybe that is a point to fix the problem

biancode, 2017-05-20:
I've searched a bit more and maxAge is not to set in:

`ClientSession.prototype.readVariableValue`

`ClientSession.prototype.readAllAttributes`

biancode, 2017-07-02:
That problem crashes Node-RED sometimes and is not to fix from outside node-opcua. I hope that is not a problem of my hardware too. The error occurs while listening/monitoring events.

erossignon, 2017-09-15:
spec says:

>  If the Server cannot meet the requested maxAge, it returns its “best effort” value rather than rejecting the request. This may occur when the time it takes the Server to process and return the new data value after it has been accessed is greater than the specified maximum age.
 > * If maxAge is set to 0, the Server shall attempt to read a new value from the data source.
 > * If maxAge is set to the max Int32 value or greater, the Server shall attempt to get a cached value.
 > * Negative values are invalid for maxAge.

erossignon, 2017-09-15:
This is strange because maxAge is tested before calling refresh in 
```
       if (request.maxAge < 0) {
            return sendError(StatusCodes.BadMaxAgeInvalid);
        }
```
so you should'nt get there. Is there any custom code that modify maxAge in the refresh callback ?

erossignon, 2018-01-26:
@biancode  is it still a issue ?

biancode, 2018-01-26:
we should close it for now - never occurred since the new 0.1.x versions

closed 2018-01-26 by biancode