node-opcua/node-opcua #348: [FEATURE] Import and export address space to JSON

pmpbar, 2017-05-03:
[FEATURE REQUEST]
```
**Expected behavior**
Ability to export address space to JSON for storage in mongo or other key value stores.
  
**What is the motivation / use case for changing the behavior?**
Having the ability to use a key value db to store and edit the address space would be pretty convenient since this is all in javascript anyways. XML doesn't really give the right storage mechanism for javascript. I have peeped at the address space code for a bit and do not think it is possible yet. 
I would be happy to implement this feature, in fact, I am offering to! I would just need some questions answered by some maintainers.
```

erossignon, 2017-05-07:
@pmpbar,
nodeset XML file are used for compatibility with other SDK ( .NET , C++ etc ), the nodeset file format is part of the OPCUA specification.
Having a JSON format to store and load address space is a great idea but I think that this should probably not be a core feature of node-opcua. I would recommend this to be added as a node-opuca extension (within a separate node-module).

pmpbar, 2017-05-08:
@erossignon, Yeah that sounds like a good idea. Keeping this as compliant as possible is important. I will start looking at making a plugin happen. You can close this if want to.

vildantursic, 2018-10-16:
Hi @erossignon ,
Does feature which creates xml from address already exists in node-opcua?
Are there any news @taybartski for that library (address to JSON), because I need same feature and I would like to join your repo and contribute if you already started working on it :)

erossignon, 2018-10-16:
@vildantursic I haven't had any news from @taybartski for the JSON module yet.
However, there are now enhanced way to produce xml file from a programatically genrated address space inside the node-opcua-address-space library.
see [https://github.com/node-opcua/node-opcua/blob/d2a1621defeb1059e76409ee631fed4f099664e1/packages/node-opcua-address-space/test/test_nodeset_to_xml.js#L399](node-opcua-address-space\test\test_nodeset_to_xml.js)

taybartski, 2018-10-16:
Sorry for not responding, we went slightly out of spec and the plugin is not owned by me now :(

vildantursic, 2018-10-17:
I was researching a bit and figured out that NodeCrawler can return nested structure, so I took it and converted it through a recursive function to JSON.
There are still few things to do like parsing a string to primitive types, and there is an issue with reading an array of objects e.g. `[ { a: 1, b: 2 } ]` where crawler returns undefined.

@erossignon can I use crawler for this parsing, or it is meant for something totally different?

I will also try Nodeset2XML and parse it back to JSON, although I am not sure if that is a good thing to do, or should I figure out another way for getting JSON from Nodeset.

I would gladly make a contribution after I finish with this project and gain enough knowledge about Node OPC UA to create this NodesetToJSON feature. :)

closed 2019-10-24 by stale[bot]