node-opcua/node-opcua #350: TypeError: Cannot read property 'cyan' of undefined

biancode, 2017-05-15:
**I'm submitting a ...**  (check one with "x")
```
[ X] bug report => search github for a similar issue or PR before submitting
[ ] feature request
[ ] support request => Please do not submit support request here, instead see use
    [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)
```

**Current behavior**
15 May 20:54:57 - [red] Uncaught Exception:
15 May 20:54:57 - TypeError: Cannot read property 'cyan' of undefined
    at /.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/client/client_base.js:339:78
    at /.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:635:28
    at Immediate.<anonymous> (/.npm-global/lib/node_modules/node-red-iiot-opcua/node_modules/node-opcua-crypto/lib/crypto_utils.js:461:9)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)
    at processImmediate [as _immediateCallback] (timers.js:617:5)

**Expected behavior**
extractPublicKeyFromCertificate from crypto_utils sends null (line 461) and client_base (line 339)
uses null.message.cyan

**Minimal reproduction of the problem with instructions**
use Basic128Rsa15 with Sign&Encrypt in server and try to connect with Basic128Rsa15 and Sign&Encrypt
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * [ ] I have installed node-opcua from source ( using git clone)
  * [ x] I have installed node-opcua as a package ( using npm install )
  * [x ] I am using an application that uses node-opcua 
        [ x] node-red
        [ ] other : please specify 
        
  * Device: _____
  * OS version: _____
    [ ] Windows : version : _________
    [ ] Linux   : version : _________
    [x ] MacOs   : version : 10.12
    [ ] Raspbian: version : _________
    [ ] Other   : specify :
    
    
* **node-opcua version:** `0.0.64`

* **Node:** `node --version` =6.10.3

erossignon, 2017-05-16:
The exception can be fixed, however the root cause (failing to extract public key: extractPublicKeyFromCertificate from crypto_utils) should also be investigated.

In order to reproduce the behavior , we will need more info about the server
Can you describe the type  of server your node-opcua is trying to connect to ?

biancode, 2017-05-16:
Dear Etienne,

constellation is a node-opcua client and server like the simple_* examples from your bin folder.
If you have a GitLab Account I can give you access to have a look.

biancode, 2017-05-16:
Dear Etienne,

I try to get a secure connection between node-opcua client and node-opcua server, but I get an error "algId and/or subjPubKey not set".
I have found the point where it comes from but it is not to see where it comes from out of node-opcua - I stopped on jsonToASN1HEX -> getEncodedHex ->getASN1Object

```
 /*
     * @since asn1x509 1.0.7
     */
    this.getASN1Object = function() {
        if (this.asn1AlgId == null || this.asn1SubjPKey == null)
            throw "algId and/or subjPubKey not set";
        var o = new KJUR.asn1.DERSequence({'array':
                                           [this.asn1AlgId, this.asn1SubjPKey]});
        return o;
    };
```

Do you have any idea, please?

Thank you!
Klaus

biancode, 2017-05-19:
"algId and/or subjPubKey not set" is a problem in using node-opcua in private NPM packages

erossignon, 2017-06-10:
May be the last 2 entries in this post need to be placed into a different issue as they do not refers to the topic of this issue: "TypeError: Cannot read property 'cyan' of undefined"

biancode, 2017-06-11:
done

biancode, 2017-06-17:
An NPM update for that would be great, because of it crashes Node-RED.

biancode, 2017-06-17:
Wow! Thank you for your great support!

closed 2017-06-17 by biancode