node-opcua/node-opcua #352: Dependency error

finalburner, 2017-05-22:
Hi everybody, 

I want to package my nodejs App (which's running without errors)  into ".exe" using the npm module "PKG".

I have theses unexplained error messages about not finding the .js component files, even if it's checking the right path.

Please, If you have any suggestion ou idea about the origin of the error, let me know.

[image]

erossignon, 2017-06-10:
This will be an issue until we switch out of the ```require("requirish")``` scheme that we use to handle relative paths.  see https://gist.github.com/branneman/8048520

finalburner, 2017-09-27:
hi @erossignon,
I've looked a while to fix the require problem and package the application event following your instructions.
The PKG module works by including all module encountred in the require calls.

Why doesn't it work as the same for node-opcua

finalburner, 2017-11-03:
Is there a way to bundle all node-opcua module in one js file in order to avoid relative path problems when packing ? @erossignon 

Thanks

erossignon, 2018-01-09:
#392confirms that packing node-opcua code into a single script file is now working thanks to the new code layout. closing

closed 2018-01-09 by erossignon