node-opcua/node-opcua #354: Installation Failed in Windows 10

subu1979, 2017-06-01:
Not able to install node-opcus in Windows 10

Node JS Version - 6.10.3
npm version - 3.10.10

steps followed
1) npm install node-opcua   -> Success
2) git clone https://github.com/node-opcua/node-opcua.git  --> Success
3) cd node-opcua
4) npm install - Failed 

[image]

erossignon, 2017-06-01:
```npm install``` has failed apparently,  you need to investigate this first. Unfortunatly the interesting bit is already passed in your screenshot

subu1979, 2017-06-01:
Hi,
Thanks for the quick reply. I re-installed and working fine.

Thanks for your support.

closed 2017-06-01 by erossignon