node-opcua/node-opcua #355: uncaughtException: AssertionError:  self.nbPendingPublishRequests >0

PiotrDeyna, 2017-06-07:
I'm submitting a 
[ X] bug report => search github for a similar issue or PR before submitting
[ ] feature request
[ ] support request => Please do not submit support request here, instead see use
    [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)
```

**Current behavior**
exception in Client:

uncaughtException: AssertionError:  self.nbPendingPublishRequests >0
    at ClientSidePublishEngine._send_publish_request (/Users/zenon/cbtrain/subscription/node_modules/node-opcua/lib/client/client_publish_engine.js:131:5)
    at Immediate.<anonymous> (/Users/zenon/cbtrain/subscription/node_modules/node-opcua/lib/client/client_publish_engine.js:85:14)
    at runCallback (timers.js:651:20)
    at tryOnImmediate (timers.js:624:5)
    at processImmediate [as _immediateCallback] (timers.js:596:5)

**Expected behavior**
no exception in Client

**Minimal reproduction of the problem with instructions**

var options = {
    connectionStrategy: {
        maxRetry: 100000
    }
};

var opcuaClient = new opcua.OPCUAClient(options);

...

with parameters:
     the_subscription = new opcua.ClientSubscription(theOpcuaSession, {
                requestedPublishingInterval: 1000,
                requestedMaxKeepAliveCount: 2,
                requestedLifetimeCount: 100,
                maxNotificationsPerPublish: 10,
                publishingEnabled: true,
                priority: 10,

goes down all the time.

with parameters:
                    the_subscription = new opcua.ClientSubscription(theOpcuaSession, {
                        requestedPublishingInterval: 10000,
                        requestedLifetimeCount: 6000,
                        requestedMaxKeepAliveCount: 2000,
                        maxNotificationsPerPublish: 1000,
                        publishingEnabled: true,
                        priority: 10
                    });

goes down every 5-10 minutes.

Number of subscribed variables: 60 which are updated max once a minute, sometimes 5-10 within a 10 ms.

**What is the motivation / use case for changing the behavior?**
I was testing with simulated server, it went down once a day.
The client tries to connect to remote server now and behaviour has changed.
The Client is running on macosx, 10.12.4
The server is running on Windows.

Client: node-opcua: 0.0.64

  * [ ] I have installed node-opcua from source ( using git clone)
  * [X ] I have installed node-opcua as a package ( using npm install )
  * [ ] I am using an application that uses node-opcua 
        [ ] node-red
        [ ] other : please specify 
        
  * Device: _____
  * OS version: _____
    [ ] Windows : version : _________
    [ ] Linux   : version : _________
    [X ] MacOs   : version : 10.12.4
    [ ] Raspbian: version : _________
    [ ] Other   : specify :
    
    
* **node-opcua version:** 0.0.64

* **Node:** `node --version` = v6.10.0

erossignon, 2017-06-10:
* Could you describe the server you're trying to connect to ?

* can you set the DEBUG variable to ```client_publish_engine```
   ```set DEBUG=client_publish_engine```
and run the server again , this will output some extra info.

PiotrDeyna, 2017-06-12:
Thanks for the answer.

I didn't get any server details but our client gives this output:

// restart here
using URSA
using URSA
ClientSidePublishEngine#registerSubscription  427
                       setting timeoutHint =  100000 100000
sending publish request 1
sending publish request 2
sending publish request 3
sending publish request 4
sending publish request 5
receive publish response
ClientSidePublishEngine.prototype._send_publish_request callback :  Cannot read property 'value' of null
'Cannot read property 'value' of null'
sending publish request 4
receive publish response
sending publish request 4
receive publish response
sending publish request 4
receive publish response
sending publish request 4
receive publish response
sending publish request 4
receive publish response
sending publish request 4

… // repeated lines

receive publish response
sending publish request 4
receive publish response
ClientSidePublishEngine.prototype._send_publish_request callback :  Cannot read property 'value' of null
'Cannot read property 'value' of null'
sending publish request 3
receive publish response
ClientSidePublishEngine.prototype._send_publish_request callback :  Cannot read property 'value' of null
'Cannot read property 'value' of null'
sending publish request 2
receive publish response
sending publish request 2
receive publish response
sending publish request 2
receive publish response
sending publish request 2
receive publish response
sending publish request 2
receive publish response
sending publish request 2
receive publish response
sending publish request 2
receive publish response
sending publish request 2
receive publish response
ClientSidePublishEngine.prototype._send_publish_request callback :  Cannot read property 'value' of null
'Cannot read property 'value' of null'
sending publish request 1
receive publish response
sending publish request 1
receive publish response
sending publish request 1

…

receive publish response
sending publish request 1
receive publish response
ClientSidePublishEngine.prototype._send_publish_request callback :  Cannot read property 'value' of null
'Cannot read property 'value' of null'
sending publish request 0

//             the client restarts here

using URSA
ClientSidePublishEngine#registerSubscription  428
                       setting timeoutHint =  100000 100000

erossignon, 2017-06-12:
We should get a steady : 
```
receive publish response
sending publish request 4
````
This exception is causing the issue:
ClientSidePublishEngine.prototype._send_publish_request callback :  Cannot read property 'value' of null
'Cannot read property 'value' of null'
sending publish request 1
```` 

I suspect that this exception is raised in one of the monitoredItem.on("change" handler.

PiotrDeyna, 2017-06-13:
the server is a KEPServerEX server. It's probably some erroneous values we are receiving.

erossignon, 2017-06-13:
In your code you should have somewhere.
```
monitoredItem.on("change",function(dataValue) {
... // some of your code here
});
```
Please add  a try/catch block around (your code here), to figure out where it's coming from

```javascript
monitoredItem.on("change",function(dataValue) {
   try {
        ... // some of your code here
   } 
   catch(err) {
       console.log("error in my code !",err);
   } 
});
```

now analyse the trace from the console.log and fix the issue
happy coding.

PiotrDeyna, 2017-06-27:
I check incoming data and improved exception handling.
Is working perfekt now, thanks !!!

closed 2017-07-11 by erossignon