node-opcua/node-opcua #356: node-opcua doe not work as part of a private package

biancode, 2017-06-11:
**I'm submitting a ...**  (check one with "x")
```
[ X] bug report => search github for a similar issue or PR before submitting
[ ] feature request
[ ] support request => Please do not submit support request here, instead see use
    [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)
```

**Current behavior**

There are problems with the security of node-opcua, when it is used in a private NPM package.

ERROR: "algId and/or subjPubKey not set".

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. install node-opcua as part of a private NPM with the '@' symbol

I have found the point where it comes from but it is not to see where it comes from out of node-opcua - I stopped on jsonToASN1HEX -> getEncodedHex ->getASN1Object

 /*
     * @since asn1x509 1.0.7
     */
    this.getASN1Object = function() {
        if (this.asn1AlgId == null || this.asn1SubjPKey == null)
            throw "algId and/or subjPubKey not set";
        var o = new KJUR.asn1.DERSequence({'array':
                                           [this.asn1AlgId, this.asn1SubjPKey]});
        return o;
    };
  
**What is the motivation / use case for changing the behavior?**

While testing and development of a package it has to be private to spread it via NPM for worldwide test partners.

**Please tell us about your environment:**

  * [ ] I have installed node-opcua from source ( using git clone)
  * [ ] I have installed node-opcua as a package ( using npm install )
  * [x ] I am using an application that uses node-opcua 
        [x ] node-red
        [ ] other : please specify 
        
  * Device: Macbook Pro
  * OS version: 
    [ ] Windows : version : _________
    [ ] Linux   : version : _________
    [ X] MacOs   : version : 10.12
    [ ] Raspbian: version : _________
    [ ] Other   : specify :
    
    
* **node-opcua version:** `0.0.64`

* **Node:** `node --version` = 6.10.3

erossignon, 2017-06-11:
Note that the exception occurs in a third partie module -> jrsasign 
Could you elaborate on "private node module" ? what do you mean.
Can you provide a concise step by step instruction to reproduce the issue on our side to ease reproducing the problem.

biancode, 2017-06-11:
In step by step, it needs a new private NPM package with a dependency to node-opcua - like the opcua-commander package - If you provide the package as private in NPM and install it from NPM, then you'll get the '@' symbol in the path. With that symbol, it does not work anymore. That is all.

erossignon, 2017-06-11:
opcua-commander is a npm package that use node-opcua as a dependency and doesn't exhibit this behavior.
we need more specific steps.. you need to explain how you create the "new private npm package with dependency" .

erossignon, 2017-06-13:
Steps that I follow:
* create a npm organisation @opcua
* use ```npm login```
* create a scope package that includes node-opcua
```
$ mkdir testscopepackage
$ cd testscopepackage
$ npm init --scope=opcua
$ npm install node-opcua --save
$ vi server.js
# ... create a simple demo server in this file ...
````

* verify that the server is up and running

```
$ node server.js 
```

* publish the package

```
$ npm publish --access public
```

* create a new projet that uses this scope pacakge
```
$cd ..
$ mkdir mytest
$ cd mytest
$ npm init
$ npm install @opcua/testscopepackage
$ vi app.js
    1.  require("@opcua/testscopepackage/server")
```

* run the app
```
$ node app.js
```

Server is working as expected .. no assert

@biancode  What is then missing ?

biancode, 2017-06-13:
Great! Now try to access via secure methods like Base128Rsa15. Please, test all (three) provided Sign or Sign and Encrypt connections. With that, you should get the problem.

biancode, 2017-06-13:
[image]

biancode, 2017-06-13:
If it works with an external client, please put the secure connecting client into that package to see what happens. I tried to connect your private package OPC UA server by the opcua-commander and it works. I have to search for the difference.

[image]

biancode, 2017-06-13:
OK, the external client works well with my package, too.

[image]

erossignon, 2017-06-13:
OK, let me know about your investigation ...

biancode, 2017-07-02:
at the end it works - my PC is broken and apparent corrupt now - I've to buy new hardware

closed 2017-07-02 by biancode