node-opcua/node-opcua #362: Error during the connection with an OPC-UA server

clebd, 2017-07-12:
**I'm submitting a ...**  (check one with "x")
```
[x] bug report => search github for a similar issue or PR before submitting
[ ] feature request
[ ] support request => Please do not submit support request here, instead see use
    [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)
```

**Current behavior**

I have an error when i'm trying to connect to an OPC-UA Server. The OPC-UA server is a High Performance sdk server from Unified Automation. I have tested with the UA Expert tool and a JAVA client and it works fine. 

```
node bin\simple_client -e opc.tcp://localhost:4840
securityMode        =  NONE
securityPolicy      =  None
timeout             =  20000
 monitoring node id =  ns=1;s=PumpSpeed
 connecting to  opc.tcp://localhost:4840
    strategy { maxRetry: 10, initialDelay: 2000, maxDelay: 10000 }
 ERROR RECEIVED FROM SENDER Bad (0x80000000)
00000000: 45 52 52 46 10 00 00 00 00 00 00 80 00 00 00 00                                                    ERRF............

 message was
undefined
 message was 2: { /*GetEndpointsRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=0
   timeStamp                   /* UtcTime                          */: 2017-07-12T14:55:50.772Z
   requestHandle               /* IntegerId                        */: 2               0x2
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 0               0x0
   additionalHeader            /* ExtensionObject                  */: null
 }
 endpointUrl                   /* String                           */:
 localeIds                     /* LocaleId                      [] */: [ /* empty*/ ]
 profileUris                   /* String                        [] */: [ /* empty*/ ]
};
 disconnected
 client : process terminated with an error
 error Error: _socket has been disconnected by third party
    at ClientTCP_transport._on_socket_ended_message (C:\Users\XXXX\node_modules\node-opcua\lib\transport\tcp_transport.js:249:18)
    at Socket.<anonymous> (C:\Users\XXXX\node_modules\node-opcua\lib\transport\tcp_transport.js:314:34)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)
 stack trace Error: _socket has been disconnected by third party
    at ClientTCP_transport._on_socket_ended_message (C:\Users\XXXX\node_modules\node-opcua\lib\transport\tcp_transport.js:249:18)
    at Socket.<anonymous> (C:\Users\XXXX\node_modules\node-opcua\lib\transport\tcp_transport.js:314:34)
    at emitNone (events.js:91:20)
    at Socket.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)
Exiting
```
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * [ ] I have installed node-opcua from source ( using git clone)
  * [X] I have installed node-opcua as a package ( using npm install )
  * [ ] I am using an application that uses node-opcua 
        [ ] node-red
        [ ] other : please specify 
        
  * Device Client : Windows 10
  * Device Server : Ubuntu 16.04

    
    
* **node-opcua version:** `aaaaf75`

* **Node:** `node --version` = v6.10.3

sasauz, 2017-07-27:
I have the same error

e-ratio, 2017-10-16:
Same error with same environment...
With node 4.2 and node-opcua 0.55 i had no error.

wesket, 2018-09-20:
What did you do to fix it?

erossignon, 2018-10-28:
no problem reported on newer version of node-opcua closing

closed 2018-10-28 by erossignon