node-opcua/node-opcua #367: How do you get the username who read a variable within refreshFunc

marcelwinh, 2017-08-30:
Hello i'm currently programming a server which gets the value of Variables from an 
external device via an async function. 
I used this example:
http://node-opcua.github.io/api_doc/classes/UAVariable.html#method_bindVariable

Variation 3:
```JavaScript
...
var options =  {
   refreshFunc : function(callback) {
     ... do_some_async_stuff_to_get_the_new_variable_value
     var dataValue = new DataValue({
         value: new Variant({...}),
         statusCode: StatusCodes.Good,
         sourceTimestamp: new Date()
     });
     callback(null,dataValue);
   }
};
...
engine.bindVariable(nodeId,options):
...
```
This works great. 
But now i have the requirement to connect to the device with the user 
which read the variable. So how do you get the information which user read 
the variable within the refreshFunction?

Currently:
USER1 => OPC UA Server => uaUser => DEVICE

Requirement:
USER1 => OPC UA Server => USER1 => DEVICE

JoshMcguigan, 2017-10-24:
I believe I know what needs to be done to add this functionality. @erossignon, is this something you would consider merging a pull request for?

erossignon, 2018-01-31:
The refresh method is used to fetch a fresh update of the uaVariable value, it doesn't depend on who's asking the variable to perform a refresh.
However, permission shall be handled at the Session level inside the opcua transaction, in the context of the current session.
@marcelwinh , can you confirm that issue #405 addessed this permission checking better and that this isssue can now be closed ?

marcelwinh, 2018-02-19:
It adresses this problem but doesn't solve it. 
The problem is the function getUserRole to check the permission is called after the refreshFunction is called. And if you are logged in as anonymous it doesn't get called at all. This could be a security Issue if the current user has no access to this node. He refreshes it's value in the server but the server then blocks the access for this user.

To reproduce:
sample_server.js
```JavaScript
var server = new opcua.OPCUAServer({
    port: 4334, // the port of the listening socket of the server
    resourcePath: "UA/MyLittleServer", // this path will be added to the endpoint resource name
     buildInfo : {
        productName: "MySampleServer1",
        buildNumber: "7658",
        buildDate: new Date(2014,5,2)
    },
    userManager: {
      isValidUser:  function (userName, password) {

        if (userName === 'test2' && password === '1234567') {
        return true;
        }
        if (userName === 'test' && password === '1234567') {
        return true;
        }
        if (userName === 'admin' && password === '1234567') {
        return true;
        }
      },

      getUserRole: function (userName) {
        console.log("getUserRole: " + new Date().getTime());
        if (userName === "admin") {
        return "admin";
        } else if (userName === "test" || userName === "test2") {
        return "operator";
        } else {
        return "guest";
        }
      },
    }
});

function post_initialize() {
    ...
    var permissionTypes = {
      CurrentRead: ['*', '!guest'],   // accept all, except guest
      CurrentWrite: ['!*', 'admin'],  // deny all except admin
    }
    var node = addressSpace.addVariable({
            componentOf: device,
            browseName: "RefreshVariable",
            dataType: "String",
      permissions: permissionTypes,
            value: new Variant({ dataType: DataType.String, value: "initValue" })
        });
        const optionsparameterAttribute = {
      refreshFunc: function (callback) {
          console.log("refreshFunction: " + new Date().getTime());
          let dataValue = new DataValue({
                      value: new Variant({ dataType: DataType.String, value: new Date().toString() }),
                      statusCode: StatusCodes.Good,
                      sourceTimestamp: new Date(),
                    });
          callback(null, dataValue);
      }
    }
    node.bindVariable(optionsparameterAttribute, true);
    ...

```
[image]

closed 2019-08-03 by stale[bot]