node-opcua/node-opcua #368: Call Method return "BadUnexpectedError"

Maestrom38, 2017-09-02:
Dear all, 

We are writing a small OPC UA client in Javascript (using the library node-opcua). The client shall simply call a Method (which takes an Int64 as argument). 

The server was developped using Python, ans seems working pretty fine. 
The Method is working properly when we use other clients like UAExpert, so the server shall be ok.

However when we execute the simple code below which:
1) Connect to the server, 
2) Create a session
3) Call the Method

We got a "BadUnexpectedError" in return and the OPC UA server seems to crash (needs to be rebooted).

Maybe I'm missing something, I copied the short code below for reference.

Thanks for your help.

`const opcua = require('node-opcua');
let client = new opcua.OPCUAClient({});

function ConnectToServer(ip,port, callback){
    let endpointUrl = "opc.tcp://" + ip + ":" + port;
    client.connect(endpointUrl,function (err) {
        if(err) {
            console.log(" cannot connect to endpoint :" , endpointUrl );
        } else {
            console.log("connected to OPC UA Serveur !");
        }
        callback(err);
    });
}

function CreateSessionCC(callback){
    client.createSession( function(err,session) {
        console.log(err);
        callback(err, session);
    });
}

function WriteHub(session, value, callback){
    let methodToCalls = [{
        objectId: "ns=3;i=1",
        methodId: "ns=3;i=50",
        inputArguments: [{dataType: DataType.Int64, value: value}]
    }];

    session.call(methodToCalls, function (err, results) {
        callback(err, results);
    });
}

ConnectToServer('192.168.0.5','4840',function(err){
   if(err){
       console.log('err: ' + err);
   }
   else {
       CreateSessionCC(function (err,session) {
           if(err){
               console.log('err: ' + err);
           }
           else {
               WriteHub(session, 2000, function(err,results){
                   if(err){
                       console.log('err: ' + err);
                   }
                   console.log('result: ' + results);
               })
           }
       })
   }
});`

erossignon, 2017-09-15:
Your code looks OK to me.
Int64 do not exist in javascript. Int64 are therefore stored as an array of two numbers, High and Low, but node-opcua shall do this for you.
The server should not crash even if a wrong parameter is passed to the method arguments.
Are you sure about 
        objectId: "ns=3;i=1",
        methodId: "ns=3;i=50"

JoelMP, 2017-09-26:
We succeeded to call the method above but only with positive values. When we try with negative value, we get this error:

`{ AssertionError: low >= 0 && low <= 0xFFFFFFFF
    at Object.exports.constructInt64 (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\misc\encode_decode.js:661:5)
    at Object.exports.coerceUInt64 (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\misc\encode_decode.js:691:20)
    at coerceVariantType (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\datamodel\variant_tools.js:69:35)
    at Object.construct_hook (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\schemas\Variant_schema.js:404:29)
    at new Variant (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\_generated_\_auto_generated_Variant.js:48:22)
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\_generated_\_auto_generated_CallMethodRequest.js:71:78    at Array.map (native)
    at new CallMethodRequest (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\_generated_\_auto_generated_CallMethodRequest.js:71:54)
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\_generated_\_auto_generated_CallRequest.js:65:76
    at Array.map (native)
    at new CallRequest (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\_generated_\_auto_generated_CallRequest.js:65:52)
    at ClientSession.call (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\client\client_session.js:970:19)
    at C:\Users\SmartHyES\Documents\PASEO\api\controllers\PilotageController.js:1159:16
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\async\dist\async.js:3853:24
    at replenish (C:\Users\SmartHyES\Documents\PASEO\node_modules\async\dist\async.js:946:17)
    at iterateeCallback (C:\Users\SmartHyES\Documents\PASEO\node_modules\async\dist\async.js:931:17)
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\async\dist\async.js:906:16
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\async\dist\async.js:3858:13
    at C:\Users\SmartHyES\Documents\PASEO\api\controllers\PilotageController.js:1155:7
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\client\client_session.js:402:9
    at C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\client\client_session.js:887:9
    at modified_callback (C:\Users\SmartHyES\Documents\PASEO\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:942:24)
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'low >= 0 && low <= 0xFFFFFFFF',
  generatedMessage: false }
`

We tried to use node modules such as node-int64 or int64-buffer but it didn’t work. We also try to give parameters value as a string but it didn’t work.

Thanks for your help.

closed 2019-08-03 by stale[bot]