node-opcua/node-opcua #369: How to resolve - FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed

awadekaustubh, 2017-09-03:
Can anyone guide me why what could be the issue which is causing the below error.

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory

How to get rid of this issue?Is it related to physical memory of my machine?

Thanks,
Kaustubh

erossignon, 2017-09-15:
see https://stackoverflow.com/questions/26094420/fatal-error-call-and-retry-last-allocation-failed-process-out-of-memory
and https://github.com/nodejs/node/issues/10137

closed 2017-10-07 by erossignon

nwaughachukwuma, 2018-01-05:
@erossignon is there a way to use **_--max_old_space_size_** with npm run unit: cross-env BABEL_ENV=test karma start test/unit/karma.conf.js --single-run? I have been trying almost all day