node-opcua/node-opcua #375: not exactly an issue, a suggestion

lucarv, 2017-09-17:
i have tried to follow the tutorial, and never got very far, i supposed due to "security issues". when following the tutorial steps there is nothing saying one should create the ssl infrastructure with certificates and private keys. 
i did that for the server (just created a selfsigned) and copied to where i think it should be, the server works.
but for the client, not so good. i can connect successfully to the server, but i can't cretae any sessions. I get: 

        applicationUri = e.tbsCertificate.extensions.subjectAltName.uniformResourceIdentifier[0];
                                                                   ^

TypeError: Cannot read property 'uniformResourceIdentifier' of undefined
    at OPCUAClient._getApplicationUri (C:\Users\lucarv\Documents\dev\node\opcua\opcua-publisher\node_modules\node-opcua-client\src\opcua_client.js:110:68)
    at OPCUAClient.__createSession_step2 (C:\Users\lucarv\Documents\dev\node\opcua\opcua-publisher\node_modules\node-opcua-client\src\opcua_client.js:178:31)
    at OPCUAClient._createSession (C:\Users\lucarv\Documents\dev\node\opcua\opcua-publisher\node_modules\node-opcua-client\src\opcua_client.js:165:10)
    at OPCUAClient.createSession (C:\Users\lucarv\Documents\dev\node\opcua\opcua-publisher\node_modules\node-opcua-client\src\opcua_client.js:584:10)
    at C:\Users\lucarv\Documents\dev\node\opcua\opcua-publisher\routes\index.js:24:18
    at Timeout.delayer [as _onTimeout] (C:\Users\lucarv\Documents\dev\node\opcua\opcua-publisher\node_modules\delayed\delayed.js:27:10)
    at ontimeout (timers.js:380:14)
    at tryOnTimeout (timers.js:244:5)
    at Timer.listOnTimeout (timers.js:214:5)

I suppose it is a cretificate issue. any hints (soory if this is a noob question)

erossignon, 2017-09-18:
The tutorial works for node-opcua<=0.65 and needs to be updated for new incoming version. Thanks for spotting this. Expecially, there is an extra step required to require the PKI.

erossignon, 2017-09-27:
This should now be working with the latest version node 0.1.9-6

closed 2017-10-07 by erossignon