node-opcua/node-opcua #376: empty EndpointUrl in RequestHeader (getEndpointsRequest)

zombinary, 2017-09-25:
I want to connect to an server as anonymous. But the server returns an error message.
The Problem is the empty endpoint Url in the GetEndpointsRequest.

Is it possible to added the EndpointUrl to the RequestHeader?

[image]

erossignon, 2017-10-07:
Not enough information provided, sorry ....

closed 2017-10-07 by erossignon

AlexeyMineev, 2018-01-22:
It seems that error is in line 661 of **client_base.js**:

**options.endpointUrl = options.endpointUrl || self.endpoint_url;**

I've replaced **self.endpoint_url** by **self.endpointUrl** and it works fine now

GhoshJit, 2018-01-25:
Read data from server is ok. but unable to write value .
Can you send me the OPC UA Client write value code in .net