node-opcua/node-opcua #377: wrong formating of an string Identifier like a GUI

zombinary, 2017-09-25:
Browse as client, a nodeId from the server which has a string Identifier like a GUI.

The Server contains a nodeId 'ns=0;s=1cf5e1fa-202a-2ab8-0440-c4fc2f22f2bf' and the
node-opcua returns a gui 'ns=0;g=1CF5E1FA-202A-2AB8-0440-C4FC2F22F2BF'.

[image]

erossignon, 2017-10-07:
Please provide more information to reproduce  or recreate the issue.

zombinary, 2017-10-09:
You can reproduce  this with an server which has an Node with the attributes
     NodeId.NamespaceIndex = 0
     NodeId.IdentifierType = String
     NodeId.Identifier = 1cf5e1fa-202a-2ab8-0440-c4fc2f22f2bf

For Example I have this Node (ns=0;s='1cf5e1fa-202a-2ab8-0440-c4fc2f22f2bf') in the ObjectFolder 'Objects' (ns=0;i=85). 
If I use the Method opcuaClient.browse('ns=0;i=85', function(err, elements){});
The array elements contains an object with the string 'element[..].nodeId = ns=0;**g**=1cf5e1fa-202a-2ab8-0440-c4fc2f22f2bf' but I'm expecting 'element[..].nodeId = ns=0;**s**=1cf5e1fa-202a-2ab8-0440-c4fc2f22f2bf'

The problem is that in this case the IdentifierType **string** is formatted to a **GUI**.

erossignon, 2017-10-11:
fixed in 0.1.0-8

closed 2017-10-11 by erossignon