node-opcua/node-opcua #378: Example code for Alarms and Events

awadekaustubh, 2017-09-26:
Could you share an example for subscribing alarms, events? 

Thanks.

erossignon, 2017-12-27:
unit tests are a good place to start.

mviikki16, 2018-12-27:
> unit tests are a good place to start.

Are the unit tests in a separate repository? Could you please share a link to them?
Thank you.

erossignon, 2018-12-27:
What is the A&E scenario you're trying to implement ? Client ? Server ? what type of alarm ?

erossignon, 2018-12-27:
For example , you could refer to this example if you want to implement an   NonExclusiveDeviationAlarm inside a server https://github.com/node-opcua/node-opcua/blob/ad79724f1934568cb1d8f84327e862bd0bad3a2b/packages/node-opcua-address-space/test/alarms_and_conditions/utest_non_exclusive_deviation_alarm.js#L33

mviikki16, 2018-12-28:
Thank you. The scenario is indeed of a server with a few variables and alarms. The source I found was    [node-opcua/packages/node-opcua-samples/bin/demo_server_with_alarm.js](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-samples/bin/demo_server_with_alarm.js) which depends on the function "construct_demo_alarm_in_address_space". 

Perhaps the ideal example would be if the [sample server](https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_server.js) would also contain an alarm for one of its variables.

srep2512, 2019-11-07:
Hey @erossignon , I know it is an old Issue. I've tried to implement the example of the test scenario and the event is triggered. 

The only problem is that my client (UAExpert) is getting an **Unexpected EventType** warning.
It would be great if you could give me a hint to define the EventType correctly, so that the client could print an readable output

```
const server = new opcua.OPCUAServer({    
    port: 4334, // the port of the listening socket of the server
    resourcePath: "/UA/MyLittleServer", // this path will be added to the endpoint resource name
    buildInfo : {
        productName: "DemoAlarmServer",
        buildNumber: "1",
        buildDate: new Date(2016,9,27)
    }
});

function post_initialize() {
    function construct_my_address_space(server) {
        as = new opcua.AddressSpace()
        const addressSpace = server.engine.addressSpace;
        const data = {};

        
        opcua.construct_demo_alarm_in_address_space(data,addressSpace);

        let time  = 1;
        function simulate_variation() {
            const value = (1.0 + Math.sin(time/360*10))/2.0;
            data.tankLevel.setValueFromSource({dataType: "Double",value:value});

            data.tankLevel2.setValueFromSource({dataType: "Double",value:value});

            time +=1;
        }
        setInterval(simulate_variation,200);
        simulate_variation();

    }
    construct_my_address_space(server);

    server.start(function() {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
        console.log(" the primary server endpoint url is ", endpointUrl );
    });
}
server.initialize(post_initialize);

```

[image]

mtbeck, 2020-04-08:
Is there a server example for using alarm condition type OffNormal?

closed 2022-02-27 by erossignon