node-opcua/node-opcua #379: CallMethod getMonitoredItems assertion error

zombinary, 2017-10-02:
I have an Problem with my client to get the Monitored Items from an server.
In the Call Method the server Response return an 'UInt32Array' with an ArraySize -1.
This hit an assertion in Line1033 in 'client_session.js'

I change the assert to Fix them. Can you apply the change?

```
                assert(result.outputArguments.length === 2);
                var data = {
                    serverHandles: result.outputArguments[0].value, //
                    clientHandles: result.outputArguments[1].value
                };
                //assert(data.serverHandles instanceof Uint32Array);
                //assert(data.clientHandles instanceof Uint32Array);
                assert(result.outputArguments[0].dataType.key === 'UInt32');
                assert(result.outputArguments[1].dataType.key === 'UInt32');
```

[image]

erossignon, 2017-10-07:
It would have been nice to fill the template:
  - what version of node-opcua ?
  - what type of server are you connecting to ?

erossignon, 2017-10-07:
The fix makes sense, when ArraySize is 0xFFFFFFF , null is returns instead of an Uint32Array,
However I would recreate 2 Uint32Array instead to make it simpler

zombinary, 2017-10-09:
I work with the node-opcua@0.0.65 and the problem occurs with an opc ua server.

closed 2017-10-09 by zombinary

erossignon, 2017-10-09:
I would have guessed that you were trying to connect to an opcua sever ;-) the question was more about which type of server : product name version etc...