node-opcua/node-opcua #380: BAD session ID Invalid ERROR

finalburner, 2017-10-07:
Hi everyone, 

Theses days I am facing a big issue with a node-opcua client with the error "BADSESSIONIDINVALID" even If I am connecting the client with an initial subscription giving the following parameters : 

  requestedPublishingInterval: 1000,
  requestedLifetimeCount: 10 ,
  requestedMaxKeepAliveCount: 2
  maxNotificationsPerPublish: 10
  publishingEnabled: true,
  priority: 10

I think that the OPC UA server is closing the session after some idle time. 

Any help would be welcome. 

Thank you

erossignon, 2017-10-09:
The sessionTimeout might be too short in your createSession.

Also I would try to increase the session requestedMaxKeepAliveCount & requestedPublishingInterval
requestedLifetimeCount: 200 ,
requestedMaxKeepAliveCount:  20

finalburner, 2017-10-10:
Thank you for you quick reply @erossignon. 

I was wondering if I could  Intercept and handle this error the restart the connexion and the session ? 

Thank you.

erossignon, 2017-10-11:
````javascript
session.on("session_closed",function(statusCode) {
    console.log("gosh ! session timeout was too short ! let's increase it");
});
```

erossignon, 2017-10-14:
I would recommand to use the built-in mechanism for reconnection:
```javascript
        var options = {
            keepSessionAlive: true,
            connectionStrategy: {
                maxRetry: 10,
                initialDelay: 2000,
                maxDelay: 10*1000
            }
        };

        client = new opcua.OPCUAClient(options);

```

finalburner, 2017-11-03:
After 5 weeks of tests. My node-opcua is still running and doesn't disconnect anymore for badSessionInvalidID.

Thanks

closed 2017-11-03 by erossignon