node-opcua/node-opcua #381: cannot find module

fuxiocteract, 2017-10-09:
**I'm submitting a ...**  (check one with "x")
```
[x ] bug report => search github for a similar issue or PR before submitting
[ ] feature request
[ ] support request => Please do not submit support request here, instead see use
    [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)
```

**Current behavior**

**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**
when running node index.js with the following code.

`const opcua = require('node-opcua');

const server = new opcua.OPCUAServer({
    port: 4334,
    resourcePath: "UA/MyLittleServer",
});

server.start(() => {
    console.log("Server is now listening ... ( press CTRL+C to stop )");
    console.log(`port ${4334}`);
});`

module.js:472
    throw err;
    ^

Error: Cannot find module '../_generated_/_auto_generated_LocalizedText'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (D:\SProj\wf-sim\node_modules\node-opcua-data-model\src\localized_text.js:8:21)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)

**Please tell us about your environment:**

  * [ ] I have installed node-opcua from source ( using git clone)
  * [x ] I have installed node-opcua as a package ( using npm install )
  * [ ] I am using an application that uses node-opcua 
        [ ] node-red
        [ ] other : please specify 
        
  * Device: PC_____
  * OS version: _____
    [x ] Windows : version :10 _________
    [ ] Linux   : version : _________
    [ ] MacOs   : version : _________
    [ ] Raspbian: version : _________
    [ ] Other   : specify :
    
    
* **node-opcua version:** `8b114b0`

* **Node:** `node --version` = 7.9.0

erossignon, 2017-10-09:
How exactly did you installed node-opcua ? can you write the exactly steps ?

KamalAman, 2017-10-25:
I am able to reproduce this issue when I install node-opcua with the `yarn` package manager. If I switch over to use NPM, it installs properly.

Maybe something is different yarns installation build steps. 

None of the _generated_ folders are created.

It would be very nice if node-opcua did support yarn.

erossignon, 2017-11-02:
This seems to be an issue with yarn, not using the same scheme to compare 2  versions numbers (semver)
 
* with npm :  '0.1.0-alpha.3f56f9b5' < '0.1.0-2' < '0.1.0-4  etc..
* with yarn :   < '0.1.0-2' < '0.1.0-4  < '0.1.0-alpha.3f56f9b5' 

as a consequence, wrong version of the module get installed with yarn.
The problem will disappear once we moved to v0.1.1 for instance and if we do not use alpha semver anymore.

yulix, 2018-01-08:
I meet the similar issue, fixed it by  upgrade node-opcua-client.

npm install node-opcua-client@0.1.1-0

erossignon, 2018-01-08:
This issue should now be fixed with incoming version 0.1.1, please confirm

erossignon, 2018-01-08:
thank you  @yulix

closed 2018-01-08 by erossignon