node-opcua/node-opcua #382: If MessageSecurityMode.NONE is disabled a client can't connect

marcelwinh, 2017-10-16:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
       [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)

**Current behavior**
If you deactivate the MessageSecurityMode.NONE the client (UaExpert) can't connect to the Server with any security option. 
The unsecure OpenSecureChannelRequest will be blocked and answered with a ServiceFault(BadSecurityPolicyRejected)
[image]

**Expected behavior**
OpenSecureChannelRequest results in an OpenSecureChannelResponse
Here i connect to a OPC UA ANSI C Demo Server v1.7.0  with disabled SecurityMode NONE.
[image]

**Minimal reproduction of the problem with instructions**

  1. use simple_server.js
  2. import: 
	const SecurityPolicy = require('lib/misc/security_policy').SecurityPolicy;
	const MessageSecurityMode = require('lib/services/get_endpoints_service').MessageSecurityMode;
  3. add security settings to options:
	securityPolicies: [SecurityPolicy.Basic128Rsa15,SecurityPolicy.Basic256],  			
    securityModes: [MessageSecurityMode.SIGN,MessageSecurityMode.SIGNANDENCRYPT],
  4. start server & try to connect
  
**What is the motivation / use case for changing the behavior?**
Allowing only secure connections to the server

**Please tell us about your environment:**

  * (x) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: Win10
    
    - (x) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:UaExpert
     - Version:1.4.4 275
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``
0.0.65

* **Node:** 
   `node --version` = v6.11.1

JoshMcguigan, 2017-10-20:
When trying to reproduce this problem, I receive the following error:

```
Cannot find module 'lib/misc/security_policy'
```

Am I missing a particular package?

erossignon, 2017-10-20:
@JoshMcguigan is using node-opcua@0.0.65 whereas you must be using the most recent version.

erossignon, 2017-10-20:
@JoshMcguigan could you verify that the first OpenSecureChannelRequest  from UAExpert doesn't ask a MessageSecuring.NONE ? (by looking into the packet)
I wonder if OPC UA ANSI C Demo Server v1.7.0 still allow connection despite the fact that SecurityMode NONE is disabled.

erossignon, 2017-10-21:
OPCUA Spec 1.03 part 4 page 10

> Discovery Endpoints shall not require any message security, but it may require transport layer
security. In production systems, Administrators may disable discovery for security reasons and
Clients shall rely on cached EndpointDescriptions. To provide support for systems with disabled
Discovery Services Clients shall allow Administrators to manually update the EndpointDescriptions
used to connect to a Server. Servers shall allow Administrators to disable the Discovery Endpoint.
>A Client shall be careful when using the information returned from a Discovery Endpoint since it
has no security. A Client does this by comparing the information returned from the Discovery
Endpoint to the information returned in the CreateSession response. A Client shall verify that:
>a) The ApplicationUri specified in the Server Certificate is the same as the ApplicationUri
provided in the EndpointDescription.
>b) The Server Certificate returned in CreateSession response is the same as the Certificate used
to create the SecureChannel.
>c) The EndpointDescriptions returned from the Discovery Endpoint are the same as the
EndpointDescriptions returned in the CreateSession response.
If the Client detects that one of the above requirements is not fulfilled, then the Client shall close
the SecureChannel and report an error.

erossignon, 2018-01-08:
@marcelwinh  could you confirm that issue is now properly resolved ?

marcelwinh, 2018-01-08:
i have tried it with the sample_server.js and it works with UaExpert.

erossignon, 2018-01-08:
Great , thank you

closed 2018-01-08 by erossignon