node-opcua/node-opcua #383: "Cannot find module" error when running simple_server demo

JoshMcguigan, 2017-10-17:
- ( x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
       [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)

**Current behavior**
After following the instructions in the readme to run the demo server, I receive the following error.

```
$ git clone https://github.com/node-opcua/node-opcua.git
$ cd node-opcua
$ npm install
$ node bin/simple_server

module.js:471
    throw err;
    ^

Error: Cannot find module '/Users/Josh/Projects/node-opcua/bin/simple_server'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:504:3
```

**Expected behavior**

After following the instructions on the readme to run the demo server, the demo server should start. 

**Minimal reproduction of the problem with instructions**
```
$ git clone https://github.com/node-opcua/node-opcua.git
$ cd node-opcua
$ npm install
$ node bin/simple_server
```

**Please tell us about your environment:**

  * ( x ) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: MacBook Pro
  * OS version:
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( x ) MacOs   : version : 10.13
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   N/A

     
    
* **node-opcua version: 7697b82faea9e9533012d6686ad7825ac0c95cdc ``

* **Node:** 
   `node --version` = v6.11.1

erossignon, 2017-10-17:
Thank you, the doc need to be updated as bin/simple_server has moved....
is it now in 
packages/node-opcua-samples/bin/simple_server.

JoshMcguigan, 2017-10-18:
That worked. Thanks for the quick response!

closed 2017-10-18 by JoshMcguigan