node-opcua/node-opcua #384: EventFilter Error after updating from 0.0.65 version

awadekaustubh, 2017-10-24:
Hi,
I have successfully instantiated the alarm. But whenever the alarm is raised I am getting following error.

```
func readValueAsync has failed
 stack TypeError: Cannot read property 'length' of null
    at extractEventField (F:\opcua\node_modules\node-opcua-service-filter\src\tools_event_filter.js:200:32)
    at Array.map (native)
```

How to resolve the issue?
Also, how to get the change event on occurrence of any event? any sample code if provided will be helpful.

erossignon, 2017-10-25:
Please fill the template and provide more information so the issue could be reproduced.

awadekaustubh, 2017-10-25:
Sorry I am new to github. Can you tell me where is the template?

erossignon, 2017-10-25:
I have pasted the template into your original post

erossignon, 2017-10-25:
@awadekaustubh  thank you for filling the template. For the record, could you also describe the client application (name, vendor, version etcc)  you're are using to connect to your node-opcua server ?
On my side I have managed to reproduce your behavior with the Prosys OPCUAClient but not with the UAExpert Client.

awadekaustubh, 2017-11-08:
@erossignon I am using Prosys OPCUAClient to test this. When can I expect this Bug fix?

closed 2018-01-08 by erossignon