node-opcua/node-opcua #390: Uncaught Exception: TypeError: Cannot read property 'derivedClientKeys' of undefined

biancode, 2017-10-30:
**I'm submitting a ...**  (check one with "x")

 - (X ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
       [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)

**Current behavior**

Uncaught Exception

**Expected behavior**

none Uncaught Exception

**Minimal reproduction of the problem with instructions**

Please, create a secure connection with Sign&Encrypt, user and password between client and server (both node-opcua) and the network (WLAN) go down for a while (short moment). It doesn't appear on insecure connections.
  
**Please tell us about your environment:**

npm 5

  * ( ) I have installed node-opcua from source ( using git clone)
  * (X ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( X) node-red
       - ( ) other : please specify 
        
  * Device: Macbook Pro
  * OS version: 10.13
    
    - ( ) Windows : version : _________
    - ( ) Linux   : version : _________
    - ( X) MacOs   : version : 10.13
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
    
* **node-opcua version:** ``

problem occures with 0.1.0-9 to 0.1.0-12

* **Node:** 
   `node --version` = 8.8.1 and 8.6.x

 ```
CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
 CANCELING TIMEOUT PUBLISH REQUEST 
Server: closing SESSION  active Session1  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:127.0.0.1  port =  51223
Server: closing SESSION  active Session1  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:127.0.0.1  port =  51233
Server: closing SESSION  active Session1  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:127.0.0.1  port =  51235
Server: closing SESSION  active Session2  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:127.0.0.1  port =  51235
ServerSidePublishEngine.transferSubscription  =< undefined
 Subscription => Notifying Transfer                                  
 ---------------------------------------------------- PUSHING PUBLISH RESPONSE FOR LATE ANWSER !
Server: closing SESSION  active Session1  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:127.0.0.1  port =  51236
Server: closing SESSION  active Session2  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:127.0.0.1  port =  51236
ServerSidePublishEngine.transferSubscription  =< undefined
 Subscription => Notifying Transfer                                  
 ---------------------------------------------------- PUSHING PUBLISH RESPONSE FOR LATE ANWSER !
 message was 
undefined
 message was 2: { /*ReadRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;b=4917a01ce35fb12759e3698ba2672e12
   timeStamp                   /* UtcTime                          */: 2017-10-29T23:20:32.046Z
   requestHandle               /* IntegerId                        */: 3055               0xbef
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: 
   timeoutHint                 /* UInt32                           */: 0               0x0
   additionalHeader            /* ExtensionObject                  */: null
 }
 maxAge                        /* Duration                         */: 0
 timestampsToReturn            /* TimestampsToReturn               */: Neither ( 3)
 nodesToRead                   /* ReadValueId                   [] */: [
   { /*0*/
     nodeId                    /* NodeId                           */: ns=0;i=2259
     attributeId               /* IntegerId                        */: 13               0xd
     indexRange                /* NumericRange                     */: NumericRange:<Empty>
     dataEncoding              /* QualifiedName                    */: <null>
   }
 ]
};
 message was 
undefined
 message was 2: <null>
 warning : ClientSessionKeepAliveManager#ping_server  _socket has been disconnected by third party
 warning : ClientSessionKeepAliveManager#ping_server  _socket has been disconnected by third party
 warning : ClientSessionKeepAliveManager#ping_server  _socket has been disconnected by third party
 warning : ClientSessionKeepAliveManager#ping_server  _socket has been disconnected by third party
 warning : ClientSessionKeepAliveManager#ping_server   ServiceResult is BadSessionIdInvalid (0x80250000)
 warning : ClientSessionKeepAliveManager#ping_server   ServiceResult is BadSessionIdInvalid (0x80250000)
30 Oct 00:20:32 - [red] Uncaught Exception:
30 Oct 00:20:32 - TypeError: Cannot read property 'derivedClientKeys' of undefined
    at ClientSecureChannelLayer._get_security_options_for_MSG (/iiot-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1196:46)
    at ClientSecureChannelLayer._sendSecureOpcUARequest (/iiot-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1229:94)
    at ClientSecureChannelLayer._internal_perform_transaction (/iiot-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1065:10)
    at ClientSecureChannelLayer._performMessageTransaction (/iiot-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:1006:10)
    at ClientSecureChannelLayer.performMessageTransaction (/iiot-opcua/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:900:10)
    at OPCUAClientBase.performMessageTransaction (/iiot-opcua/node_modules/node-opcua-client/src/client_base.js:602:25)
    at ClientSession.performMessageTransaction (/iiot-opcua/node_modules/node-opcua-client/src/client_session.js:872:18)
    at ClientSession._defaultRequest (/iiot-opcua/node_modules/node-opcua-client/src/client_session.js:579:10)
    at ClientSession.publish (/iiot-opcua/node_modules/node-opcua-client/src/client_session.js:750:10)
    at ClientSidePublishEngine._send_publish_request (/iiot-opcua/node_modules/node-opcua-client/src/client_publish_engine.js:140:18)
    at Immediate._onImmediate (/iiot-opcua/node_modules/node-opcua-client/src/client_publish_engine.js:84:14)
    at runCallback (timers.js:785:20)
    at tryOnImmediate (timers.js:747:5)
    at processImmediate [as _immediateCallback] (timers.js:718:5)

```

erossignon, 2017-10-30:
Please also fill
 * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://

erossignon, 2017-10-30:
OK,I can reproduce:

biancode, 2017-10-30:
in that case I just used node-opcua - there was no other OPCUA system

closed 2018-01-08 by erossignon

samminen, 2018-11-16:
is this issue resolved ? I am also facing server session  timeout issue; 
*********************************************************
initialized
Server is now listening ... ( press CTRL+C to stop)
port  4334
 the primary server endpoint url is  opc.tcp://NTBK579005.CORP.HALLIBURTON.COM:4334/UA/MyLittleServer
Server: closing SESSION  active Session1  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:34.52.44.62  port =  57430
Server: closing SESSION  active Session1  because of timeout =  60000  has expired without a keep alive
channel =  ::ffff:34.52.44.62  port =  57441

cleancoderocker, 2020-08-07:
Is it possible to set this timeout on the OPC UA server provided by the node-opcua library?

StefanoDalMas, 2023-11-03:
> Is it possible to set this timeout on the OPC UA server provided by the node-opcua library?

Did you find any solution to this? I'm having the same issue