node-opcua/node-opcua #392: relative paths, requirish, browserify

finalburner, 2017-11-03:
Hi, 

As I'm still investigating a way to pack the node-opcua module with my node.js application.
see **https://github.com/node-opcua/node-opcua/issues/352**

I had already packaged many Node.js Application specially using Pkg or Nexe.

But here, in order to package node-opcua, I found myself with relatif paths problems systematically.

So, instead of packing all the module with my application, I've thought that I could bundle the module , which means that I could make all the module in one .js file

I have tried with many npm bundling modules : bundle, bundle-js, browsify, browserify.

Actually, I think that browserify could do the job. Now I found that browserify can do some 
 transformation using requirish ( which is the module used by node-opcua to avoid relatif path problems example : ./../../.. ), so that it could rewrite the complex relative paths.

See bottom of the page : https://www.npmjs.com/package/requirish

So I executed this command : 

browserify -t requirish node_modules/node-opcua/index.js  > bundle.js

And still I encountred some errors finding some modules.

ANY help would be welcome.

Great thanks for your interest.

erossignon, 2017-11-03:
please use most recent version of node-opcua,as they do not use requirish anymore.

finalburner, 2017-11-03:
In fact, as I update to 0.1.0-12, I discovered a major change in module organisation. After that, I just packaged my application using PKG. It was sucessful.

Great Thank.

closed 2017-11-03 by erossignon