node-opcua/node-opcua #396: installation failed while generating self-signed certificate

giminni, 2017-11-17:
I tried to install node-red-iiot-opcua under Docker and Alpine Linux.
Alpine Linux uses libreSSL and there is no -rand option available.

Therefore the installation was partial successful, because could not build the
needed certificates due to the lack of -rand in openssl.

The connection to an OPC/UA server will fail later due to the lack of certificates.

Can the certificate be build without using a random file?

    OpenSSL version : LibreSSL 2.6.3
    ########################################### OPENSSL ERROR ###############################
    Command failed: "openssl" genrsa -out "/usr/src/node-red/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/usr/src/node-red/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
    usage: genrsa [args] [numbits]
    -des encrypt the generated key with DES in cbc mode
    -des3 encrypt the generated key with DES in ede cbc mode (168 bit key)
    -idea encrypt the generated key with IDEA in cbc mode
    -aes128, -aes192, -aes256
    encrypt PEM output with cbc aes
    -camellia128, -camellia192, -camellia256
    encrypt PEM output with cbc camellia
    -out file output the key to 'file
    -passout arg output file pass phrase source
    -f4 use F4 (0x10001) for the E value
    -3 use 3 for the E value

########################################### OPENSSL ERROR ###############################

erossignon, 2017-11-17:
with alpine linux1 4.4.59  I have 

    $ openssl version
    OpenSSL 1.0.2m 2 Nov 2017 (Library: OpenSSL: 1.0.2k 26 Jan 2017)

and 
    openssl genrsa --rand seems to be working

Can you confirm the version of Alpine Linux used on your end and openssl version 
Have you tried updating the Alpine Linux image to the latest ?

giminni, 2017-11-17:
Hi Etienne,
yes indeed I am on alpine linux edge, I'll sent you my configuration details later.

erossignon, 2017-11-17:
It looks that there are two flavors of openssl  LibreSSL and ... OpenSSL.
```apk add libressl```  and ```apk add openssl```  both install an openssl command in   /usr/bin/openssl
Obviously the two conflict.
Please try to reinstall the genuine openssl version :  ```apk add openssl```

giminni, 2017-11-17:
Hi Etienne,

thanks for pointing me to use the openssl package.

I tested it and yes it works on alpine edge.

I thought Alpine Linux has switched to libreSSL on edge.
See [Alpine edge has switched to libressl](http://lists.alpinelinux.org/alpine-devel/5463.html)

closed 2017-11-17 by giminni

mikakaraila, 2017-11-18:
Ack, I was just busy & travelling during last week. Good that solution was found.

giminni, 2017-11-19:
For all just to know, Alpine LInux exchanged in the past openSSL with libreSSL.
Now they use for Alpine Linux core libressl-libs and userland can stuck with openssl.
IMHO if openSSL is weak in so many aspect why not switch to libreSSL.