node-opcua/node-opcua #398: Making node-opcua more portable

fabiopelosin, 2017-11-21:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - (x) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
       [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)

Hello and thank you for the awesome library. We are investigating the usage of node-opcua for a client in an environment different from node were some of the node modules are not available.

Initials test have been positive however, the dependence on the `fs` module doesn’t allow to complete sucessfully a client connection. Our preliminary investigation revealed that access to the file system is mostly used to load certificates. If there is interest in this feature we are interested in making a pull request to intruduce support for different strategies for the features relying on a node module.

In any case any guidance in this regard would be very helpful.

Thank you and have a great day.

jakobgruber, 2018-01-12:
Hello Fabio,
did you successfully run node-opcua in a different environment? I am interested in this feature because I would like to use opcua on android via ionic / cordova.
Best regards, Jakob

antoniuschan99, 2018-01-17:
Hey I am able to get it running on node-js-mobile by Janaesystems.

erossignon, 2018-01-31:
see also #263

closed 2019-08-03 by stale[bot]