node-opcua/node-opcua #401: Connect to OPC DA server

helderburato, 2017-11-25:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report
 - ( ) feature request
 - (x) support request
  
**Current behavior**
I'm trying to connect in **OPC DA**, my question is if it's possible to connect in **OPC DA** server using this **OPC UA** module, anyone has used for this case?

**Please tell us about your environment:**
I'm creating client to connect in server `OPC DA`.

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
        
  * Device: VM
  * OS version: Windows 7 Ultimate
    
    - (x) Windows : version : WinCC Server
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: Siemens OPC
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://

* **Node:** 
   `node --version` = v8.9.1 LTS

allatrue, 2017-11-26:
OPC DA is a classic OPC based on DCOM, so per definition windows only. 
See https://en.m.wikipedia.org/wiki/OPC_Unified_Architecture

> Although developed by the same organization, OPC UA differs significantly from its predecessor, Open Platform Communications (OPC). The Foundation's goal for OPC-UA was to provide a path forward from the original OPC communications model (namely the Microsoft Windows-only process exchange COM/DCOM) that would better meet the emerging needs of industrial automation.

eoursel, 2017-11-26:
Yes but you can use an OPC DA to OPC UA converter on Windows. With this component you can have a node opcua client talking to an OPC DA server. There are several converters on the market sold by some opc foundation vendors. 

Regards
⁣Envoyé par BlueMail ​

helderburato, 2017-12-05:
Thank you all!

It was necessary to purchase the OPC DA license to connect on **WIN CC Server OPC UA** using **OPC UA** package.

I bought her and the problem was solved.

closed 2017-12-05 by helderburato

shrutifiske, 2018-11-28:
Hi @helderburato  
Did you use any converter like wrapper/proxy or only OPC DA license work for you?