node-opcua/node-opcua #402: SamplingInterval or PublishingInterval have no effect : minimum 1 second

mathieu-gouhier, 2017-11-28:
**I'm submitting a ...**  (check one with "x")

 - (*) bug report => search github for a similar issue or PR before submitting
 - () feature request
 - ( ) support request => Please do not submit support request here, instead see use 
       [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)

**Current behavior**

Subscription publish a value every one second with this parameters :

requestedPublishingInterval: 100,
requestedLifetimeCount: 10,
requestedMaxKeepAliveCount: 2,
maxNotificationsPerPublish: 10,
publishingEnabled: true,
priority: 10

samplingInterval: 100,
discardOldest: true,
queueSize: 10

For information, the node value (server side) change every ms.

**Expected behavior**

Change the publish comportment : possibility to adjust at 100ms, 10ms or 1ms

**Minimal reproduction of the problem with instructions**

  1. ________________
  2. ________________
  3. ________________
  ...
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

OS : Windows 7 or Linux alpine
OPCUA Server : node-opcua
Test : UAexpert

  * ( ) I have installed node-opcua from source ( using git clone)
  * (*) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - (*) Windows : version : 7
    - (*) Linux   : version : alpine
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``

A new installation was made yesterday with "npm install" command

* **Node:** 
   `node --version` = 6.11.3

erossignon, 2017-12-27:
Can you provide an example to reproduce the behaviour you have ?

Remember that the sampling Interval will be minored by the value of the minimumSampling interval specified on the UAVariable you're trying to monitor, regardless of how often the server update the value internally. This may explain why you have an update every 1000ms .

mathieu-gouhier, 2018-01-19:
The problem was solved since we have a new opc ua server. So it was not a client side problem.

Thanks for your help.

erossignon, 2018-01-20:
Closing then

closed 2018-01-20 by erossignon

marracash, 2018-09-12:
Hi, I've created a server (with this library) with an attribute that has a variable update rate. If I use a 100 ms rate all work, but if I use a 50 (or less) rate the monitoring get the update with a variable jitter (often as a burst): is this a limit or I can tune some parameters in order to solve this?

These are my params:
_requestedPublishingInterval:0
requestedLifetimeCount:1000
requestedMaxKeepAliveCount:5
maxNotificationsPerPublish:100
publishingEnabled:true
priority:10
samplingInterval:0
queueSize:10000
discardOldest:false_

In order to simplify my request

- if I have this opcua update sequence of attribute 
           _value:1....[10ms]....value:3.....[10ms].....value:2....[10ms]...etc._
- I wish have in the monitoring the same sequence with the same jitter time, instead I have
           _value:1....[6ms]....value:3.....[0ms]....value:2...[12ms]...etc._

I have this problem only under 100 ms...I'm in a local network (no internet).

Thanks in advance

PietroGreco, 2019-07-05:
Hi,
I have a similar problem. The server (written with this library) sets a lower bound on samplingInterval equal to 50ms.
Searching in the code I have found this line (packages/node-opcua-server/source/monitored_item.ts):

https://github.com/node-opcua/node-opcua/blob/374513ddb17abde218c06f629f16038e378dee63/packages/node-opcua-server/source/monitored_item.ts#L309

Changing 50 to 1, I am able to request lower sampling intervals.
I do not know the other parts of the code but, what I am missing?