node-opcua/node-opcua #412: Running on Mobile

antoniuschan99, 2018-01-13:
Hi, I'm starting a project to use OPC UA for a controller to link with a Mobile Device. Planning on using React Native and somehow run nodejs to extra and jog data to the controller. A simple POC works on desktop running nodejs, and now going to try to run it on mobile.

Have looked into this to run nodejs: https://github.com/janeasystems/nodejs-mobile

Is it even needed to run nodejs? 

Thank you!

erossignon, 2018-01-13:
Did you known that node-opcua can run on termux ?
https://github.com/node-opcua/node-opcua/wiki/installing-node-opcua-on-your-Android-Phone-with-Termux
Let us know about about your finding to use it on Android using an other strategy ...

antoniuschan99, 2018-01-13:
Awesome, I did not see this!

Termux works for Android, how to get it running on iOS?

antoniuschan99, 2018-01-16:
I was able to run it on iOS using the Janeasystems Nodejs Mobile Library :)

closed 2018-01-16 by antoniuschan99

RakshanPremsagar, 2019-05-15:
How to make an android app with nodejs?