node-opcua/node-opcua #414: server_with_method example does not work

korny86, 2018-01-16:
**I'm submitting a ...**  (check one with "x")

 - (x ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

I get an assertion error when I call the method with the OPC UA client.

ERR in method  handler options.arrayType === VariantArrayType.Matrix, "dimension  can only provided if variant is a matrix"
AssertionError [ERR_ASSERTION]: options.arrayType === VariantArrayType.Matrix, "dimension can only provided if variant is a matrix"

**Expected behavior**

Call without assertion
**Minimal reproduction of the problem with instructions**

`var printer3DControl = addressSpace.addObject({ organizedBy: addressSpace.rootFolder.objects, browseName: "UM3Controller" });
    
    /* ---------- PAUSE -------------*/
    var Pause = addressSpace.addMethod(printer3DControl, {

        browseName: "Pause",

        inputArguments: [ ],

        outputArguments: [{
            name: "PauseResult",
            description: {text: "Boolean result of method"},
            dataType: opcua.DataType.String,
            valueRank: 1
        }]
    });
    
    Pause.bindMethod(function (inputArguments, context, callback) {

    console.log("I try to pause the print!");
        
        var returnValue = [];
        returnValue.push("Neiiiiiin");
        
        var callMethodResult = {
            statusCode: opcua.StatusCodes.Good,
            outputArguments: [{
                dataType: opcua.DataType.String,
                arrayType: opcua.VariantArrayType.Array,
                dimensions: [returnValue.length],
                value: returnValue
            }]
        };
        callback(null, callMethodResult);
    });`
  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: _____
    
    - ( x) Windows : version : 7
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
    Server: node.js with node-opcua
    Client: UA Expert
    
* **node-opcua version:** ``

`-- node-opcua@0.1.1-0

* **Node:** 
   `node --version` = v8.9.4

erossignon, 2018-01-17:
Don't shoot on the ambulance !
The assertion is providing a relevant message that can help you fixing the issue in your code.
```AssertionError [ERR_ASSERTION]: options.arrayType === VariantArrayType.Matrix, "dimension can only provided if variant is a matrix"```

so 
```javascript
        outputArguments: [{
            dataType: opcua.DataType.String,
            arrayType: opcua.VariantArrayType.Array,// <= this is an Array , not a MATRIX
            ////// REMOVE THIS LINE dimensions: [returnValue.length],
            value: returnValue
        }]
```

erossignon, 2018-01-17:
thank you @korny86  the bug was introduced recently and will be fixed in next release.

closed 2018-01-17 by erossignon

korny86, 2018-01-17:
Thanks for your quick help! *thumbs up*