node-opcua/node-opcua #415: Writing

antoniuschan99, 2018-01-16:
I am having trouble writing to the OPC UA server (it's using CodeSys). I'm using the sample client (the one with callbacks). The write seems to be going through, but the value never changes. Here is the console:

free mem % =  DataValue {
  value: 
   Variant {
     dataType: { [Number: 10] key: 'Float', value: 10 },
     arrayType: { [Number: 0] key: 'Scalar', value: 0 },
     value: 100,
     dimensions: null },
  statusCode: { [Number: 0] value: 0, description: 'No Error', name: 'Good' },
  sourceTimestamp: 2000-01-05T08:45:40.470Z,
  sourcePicoseconds: 0,
  serverTimestamp: 2000-01-05T08:45:40.470Z,
  serverPicoseconds: 0 }
-------------------------------------
subscription started for 2 seconds - subscriptionId= 2
 % free mem =  DataValue {
  value: 
   Variant {
     dataType: { [Number: 10] key: 'Float', value: 10 },
     arrayType: { [Number: 0] key: 'Scalar', value: 0 },
     value: 0.8399999737739563,
     dimensions: null },

And here is the code:

 var nodesToWrite = [
            {
                nodeId: "ns=2;s=|var|Controller.Application.GVL_PickAndPlace.OPC_UA.rYAxisActualPosition",
                attributeId: opcua.AttributeIds.Value,
                indexRange: null,
                value: { /* dataValue*/
                    serverTimestamp: new Date(),
                    serverPicoseconds: 20,
                    sourceTimestamp: new Date(),
                    sourcePicoseconds: 30,
                    value: { /* Variant */
                        dataType: opcua.DataType.Float,
                        value: parseInt(100)
                    }
                }
            }
        ];

        the_session.write(nodesToWrite, function(err,statusCode,diagnosticInfo) {
            if (!err) {
                console.log(" write ok" );
                console.log(statusCode);
                console.log(diagnosticInfo);
            }
        }); 
        callback();

erossignon, 2018-01-17:
Please use the appropriate forum for nodejs question and use the template form

erossignon, 2018-01-17:
Callback is misplaced:
```` javascript
        the_session.write(nodesToWrite, function(err,statusCode,diagnosticInfo) {
            if (!err) {
                console.log(" write ok" );
                console.log(statusCode);
                console.log(diagnosticInfo);
            }
           callback();  // should be here !!
        }); 
```

antoniuschan99, 2018-01-17:
ok sorry will close this issue

closed 2018-01-17 by antoniuschan99