node-opcua/node-opcua #419: minimumSamplingInterval of variables has no effect

marcelwinh, 2018-01-29:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
       [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua)

**Current behavior**
minimumSamplingInterval of a variable has no effect. If you create a variable with a minimumSamplingInterval of 1000ms and you view it with UaExpert, the minimumSamplingInterval is displayed correctly. If you monitor it UaExpert also takes the correct sampling interval of 1000ms but if you change that interval to 100ms it succeeds and the variable is refreshed every 100ms.
[image]
[image]

**Expected behavior**
You get as response a revisedSamplingInterval of the minimumSamplingInterval of the variable
[image]

**Minimal reproduction of the problem with instructions**

  1. use simple_server.js
  2. add ```minimumSamplingInterval: 1000``` to a variable
  3. monitor it with UaExpert and change the sampling interval
  
**What is the motivation / use case for changing the behavior?**
The server should block/change invalid sampling rates

**Please tell us about your environment:**

  * (x) I have installed node-opcua from source ( using git clone)
  * ( ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
  
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: PC
  * OS version: Win10
    
    - (x) Windows : version : 10
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:UaExpert
     - Version:1.4.4 275
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:** ``
0.0.65 & 0.2.0

* **Node:** 
   `node --version` = v6.11.1

in "packages/node-opcua-server/src/monitored_item.js"
Line 945:
```
MonitoredItem.prototype.modify = function (timestampsToReturn, options) {
    ...
L955: 
    if (old_samplingInterval !== 0 && options.samplingInterval === 0) {
        options.samplingInterval = minimumSamplingInterval; // fastest possible
    }
```
here you check if the given sampling interval is "0" and set it to the server minimumSamplingInterval and not to the minimumSamplingInterval of the variable which might be higher.

my small workaround would be:
```
L955:
       if (old_samplingInterval !== 0 && options.samplingInterval === 0) {
		if (self._node.minimumSamplingInterval > minimumSamplingInterval) {
		  options.samplingInterval = self._node.minimumSamplingInterval;
		} else {
		  options.samplingInterval = minimumSamplingInterval; // fastest possible
		}
	} else {
		if(self._node.minimumSamplingInterval > options.samplingInterval) {
		  options.samplingInterval = self._node.minimumSamplingInterval;
		}
	}
```
but i don't know if this is the right place...

erossignon, 2018-01-30:
Note that it is not always possible to limit the requestedSamplingInterval with the node minimumSamplingInterval, as the node may not exist at the time when the monitored item is created ( either unmet remote node or inexistant node in the local addressSpace)

I suppose that the minimumSamplingInterval on the UAVariable is a indication for the client of the suitable sampling interval rather than an enforced limit.

We will need to cross-checked inside the spec to find the expected behavior

marcelwinh, 2018-01-31:
You are right, i didn't look at that case. But then it should be:
If the variable exists take the highest sampling interval of variable & server
If the variable does not exist return BadNodeIdUnknown (as it is right now)
Here is the spec (Part 4 R1.04 p. 61): 
[image]

erossignon, 2018-01-31:
@marcelwinh could you check if the improvement proposed in commit 20358bc is fixing the issue to your satisfaction ?

marcelwinh, 2018-01-31:
i have checked it and it works as expected

closed 2018-01-31 by marcelwinh