node-opcua/node-opcua #420: await session.read() for single node returns error in 0.2.0

coussej, 2018-01-29:
**I'm submitting a ...**  (check one with "x")

 - (x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request
 - ( ) support request => Please do not submit support request here, instead see use 
      - use [gitter](https://gitter.im/node-opcua/node-opcua) or [stackoverflow](https://stackoverflow.com/questions/tagged/node-opcua) for community support, or 
      - [contact Sterfive for commercial support](https://www.sterfive.com)

**Current behavior**

When performing a read on a single node (as described [here](http://node-opcua.github.io/tutorial/2018/01/22/node-opcua-embraces-async-await.html), an error is thrown.

**Expected behavior**

A value is returned.

**Minimal reproduction of the problem with instructions**
```javascript
const opcua = require('node-opcua')

let endpointUrl = 'opc.tcp://localhost:53530/OPCUA/SimulationServer'
let client = new opcua.OPCUAClient({
  connectionStrategy: {
    maxRetry: 3,
    initialDelay: 1000,
    maxDelay: 10000
  },
  keepSessionAlive: true
})

;(async () => {
  await client.connect(endpointUrl)
  let session = await client.createSession()

  let datavalues = await session.read([{ nodeId: 'ns=5;s=Sinusoid1' }]) // works as expected
  console.log(datavalues)

  let datavalue = await session.read({ nodeId: 'ns=5;s=Sinusoid1' }) // throws an error
  console.log(datavalue)
})()
```

**Please tell us about your environment:**

MacOS 10.13
 
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:Prosys Simulation OPCUA Server
     - Version:2.3.2
     - Manufacturer/Software vendor: Prosys
     
    
* **node-opcua version:** ``
0.2.0

* **Node:** 
   `node --version` = v8.9.0

erossignon, 2018-01-30:
Will be fix in next minor release 
the following script works with the master branch
``` javascript
let endpointUrl = 'opc.tcp://opcuademo.sterfive.com:26543'
let client = new opcua.OPCUAClient({
  connectionStrategy: {
    maxRetry: 3,
    initialDelay: 1000,
    maxDelay: 10000
  },
  keepSessionAlive: true
})

;(async () => {
  await client.connect(endpointUrl)
  let session = await client.createSession()

  let datavalues = await session.read([{ nodeId: 'ns=1;s=Temperature' }]) // works as expected
  console.log(datavalues[0].toString())

  let datavalue = await session.read({ nodeId: 'ns=1;s=Temperature' }) // throws an error
  console.log(datavalue.toString())

  await session.close();
  await client.disconnect();
})();

```

coussej, 2018-01-30:
Great ! I just realised I didn't include the actual error message, for which my apologies! But I'm glad you were able to fix it anyway 🎉

erossignon, 2018-01-31:
Jeroen , can you test that this now fixed with version 0.2.1 ?

coussej, 2018-02-01:
@erossignon With the example program above I still get the error. I'll include it this time:

```
ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property '0' of null
    at ./node_modules/node-opcua-client/src/client_session.js:804:96
    at ./node_modules/node-opcua-client/src/client_session.js:1157:9
    at modified_callback (./node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:986:28)
    at process_request_callback (./node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:76:5)
    at ClientSecureChannelLayer._on_message_received (./node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:122:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (./node_modules/node-opcua-secure-channel/src/message_builder.js:442:22)
    at MessageBuilderBase._feed_messageChunk (./node_modules/node-opcua-transport/src/message_builder_base.js:192:18)
    at PacketAssembler.<anonymous> (./node_modules/node-opcua-transport/src/message_builder_base.js:52:14)
(node:13882) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property '0' of null
(node:13882) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

The error originates from line 804 where callback(...) is called, but I can't figure out what the difference is between the read() call with or without an array. From line [764](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/src/client_session.js#L764) onwards, the logic seems to be the same for both cases?

fossejc, 2018-02-10:
Hi all,

I have the same issue with the same code, node-opcua version (0.2.1), nodejs version (v8.9.4) and MacOS 10.13.2.

Any ideas on how to solve it ?

Best,

erossignon, 2018-02-10:
please provide the code that call ```session.read``` with enough context for investigation

fossejc, 2018-02-10:
Thank you for your answer. 

I was using the following code:
```
let endpointUrl = 'opc.tcp://opcuademo.sterfive.com:26543'
let client = new opcua.OPCUAClient({
  connectionStrategy: {
    maxRetry: 3,
    initialDelay: 1000,
    maxDelay: 10000
  },
  keepSessionAlive: true
})

;(async () => {
  await client.connect(endpointUrl)
  let session = await client.createSession()

  let dataValues = await session.read([{ nodeId: 'ns=1;s=Temperature' }]) // works as expected
  console.log(dataValues[0].toString())

  let dataValue = await session.read({ nodeId: 'ns=1;s=Temperature' }) // DO NOT throws an error
  console.log(dataValue.toString())

  await session.close();
  await client.disconnect();
})();
```

I just tried to remove these lines and it seems to work now.
```
let dataValue = await session.read({ nodeId: 'ns=1;s=Temperature' }) // DO NOT throws an error
console.log(dataValue.toString())
```

Best,

erossignon, 2018-02-11:
@fossejc , please double check your installed version of node-opcua. it should 0.2.1 for this code to work.
(I've tested it again on my side). You may have to delete your package-lock.json  file or regenerate  your node_module folder

erossignon, 2018-02-11:
@coussej : the exception that you have is usually raised from within a callback function passed to a node-opcua async method. 
Have you adapted your code to the recent API change on the ClientSession#read api (see [release note](https://github.com/node-opcua/node-opcua/releases/tag/v0.2.0) 
Please rerun your programm with the DEBUG env set to API.
This will display a warning to help you spot places where you need to adjust your code.

fossejc, 2018-02-12:
@erossignon I made sure to run version 0.2.1 but I still got an error if I keep these lines
```
let dataValue = await session.read({ nodeId: 'ns=1;s=Temperature' }) // DO NOT throws an error
console.log(dataValue.toString())
```

Here is the full output:
```
DataValue:
   value:           Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: Cannot read property '0' of null
    at /Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-client/src/client_session.js:804:96
    at /Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-client/src/client_session.js:1157:9
    at modified_callback (/Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:986:28)
    at process_request_callback (/Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:76:5)
    at ClientSecureChannelLayer._on_message_received (/Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-secure-channel/src/client/client_secure_channel_layer.js:122:5)
    at emitMany (events.js:147:13)
    at MessageBuilder.emit (events.js:224:7)
    at MessageBuilder._decode_message_body (/Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-secure-channel/src/message_builder.js:442:22)
    at MessageBuilderBase._feed_messageChunk (/Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-transport/src/message_builder_base.js:192:18)
    at PacketAssembler.<anonymous> (/Users/fossejc/Desktop/Developments/OpcClient/node_modules/node-opcua-transport/src/message_builder_base.js:52:14)
(node:5380) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property '0' of null
(node:5380) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

coussej, 2018-02-12:
Hi @erossignon, 

Just ran it with the debug variable set to API, and I indeed get messages that the API has changed. However, I'm not calling the async method with a callback. I'm using the exact code as in the initial issue description (and in the blog): `await session.read({ nodeId: 'ns=5;s=Sinusoid1' })`. 

I think it might have something to do with these lines in ClientSession.prototype.read() [L749](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/src/client_session.js#L749) :

```javascript
    if (!callback) {
        callback = maxAge;
        maxAge = 0;
    }
```

erossignon, 2018-02-14:
OK Found it ! I manage to reproduce with Prosys Demo server. The issue comes from diagnostic info to nul in the response string.
It is already fixed in master branch , fix will be in next pacth

erossignon, 2018-02-28:
@coussej  @fossejc  could you confirm this is OK now ?

coussej, 2018-03-01:
@erossignon OK! Just ran the code with v.0.2.2 and I now get the expected behaviour.

fossejc, 2018-03-02:
@erossignon It is working fine with version 0.2.2
Thanks !

closed 2018-03-02 by erossignon